NVIDIA-AI-IOT/redtail #2: Probems building sim code

griz1112, 2017-09-11:
Trying to build the ros packages I get this error on px4_controller build. 
/root/redtail/ros/packages/px4_controller/src/px4_controller.cpp: In member function 'bool px4_control::PX4Controller::parseArguments(const ros::NodeHandle&)':
/root/redtail/ros/packages/px4_controller/src/px4_controller.cpp:287:20: error: 'make_unique' is not a member of 'std'
         vehicle_ = std::make_unique<Drone>();

Any ideas ?

griz1112, 2017-09-11:
Reinstalled the compiler upgrade and it works now. Fatfingered the update-alternatives line. Great work thanks for this package.

Alexey-Kamenev, 2017-09-11:
Thank you for the feedback! 
We tried first to "dockerize" simulator instructions but found it sometimes being too brittle so decided to post the steps as-is. Releasing that Dockerfile can still be a good idea though.

I'm closing the issue, feel free to re-open/create a new one if needed.

closed 2017-09-11 by Alexey-Kamenev