NVIDIA-AI-IOT/redtail #3: ArduPilot Support

auturgy, 2017-09-12:
This looks impressive, but lacks support for the largest potential user and developer base.

Alexey-Kamenev, 2017-09-12:
Thank you for the feedback!
We do not have immediate plans to support APM stack, but it should be easy to do: our controller relies on MAVROS/MAVLINK so you only need to change corresponding control commands to support APM.

auturgy, 2017-09-12:
Thanks for the quick response.
I hope you review your plans re ArduPilot. I'd be interested to understand your reasoning: you're abstracted from the code, so licensing isn't an issue.

griz1112, 2017-10-14:
I'm working with the Ardupilot folks to firm up the support for ROS. The fact is apm hasn't been all that great until recently supporting ROS. This has changed especially with the Rover code. The ArduRover came in today so I'll be starting to test this next week. I have a 3 camera rig for it that is calibrated and ready to gather data.

fnoop, 2017-12-29:
Is this currently unsupported for arducopter?  As in it uses ROS commands only supported by PX4 firmware, for now?