NVIDIA-AI-IOT/redtail #5: Invoking "make px4_controller_node -j4 -l4" failed

TritonSailor, 2017-09-21:
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/ws/build"
####
####
#### Running command: "make px4_controller_node -j4 -l4" in "/home/ubuntu/ws/build"
####
[ 33%] Building CXX object px4_controller/CMakeFiles/px4_controller_node.dir/src/px4_controller.cpp.o
/home/ubuntu/ws/src/px4_controller/src/px4_controller.cpp: In member function ‘bool px4_control::PX4Controller::arm()’:
/home/ubuntu/ws/src/px4_controller/src/px4_controller.cpp:531:77: error: ‘mavros_msgs::SetMode::Response {aka struct mavros_msgs::SetModeResponse_<std::allocator<void> >}’ has no member named ‘success’
             if (setmode_client_.call(offb_setmode) && offb_setmode.response.success)
                                                                             ^
px4_controller/CMakeFiles/px4_controller_node.dir/build.make:62: recipe for target 'px4_controller/CMakeFiles/px4_controller_node.dir/src/px4_controller.cpp.o' failed
make[3]: *** [px4_controller/CMakeFiles/px4_controller_node.dir/src/px4_controller.cpp.o] Error 1
CMakeFiles/Makefile2:4014: recipe for target 'px4_controller/CMakeFiles/px4_controller_node.dir/all' failed
make[2]: *** [px4_controller/CMakeFiles/px4_controller_node.dir/all] Error 2
CMakeFiles/Makefile2:4026: recipe for target 'px4_controller/CMakeFiles/px4_controller_node.dir/rule' failed
make[1]: *** [px4_controller/CMakeFiles/px4_controller_node.dir/rule] Error 2
Makefile:1603: recipe for target 'px4_controller_node' failed
make: *** [px4_controller_node] Error 2
Invoking "make px4_controller_node -j4 -l4" failed
All done.

Alexey-Kamenev, 2017-09-21:
Is this while running `jetson_ros_install.sh` script? Judging by the paths, it looks like you are running it on TX-1, is that correct? Basically, a bit more details would be nice so we can try to repro.

TritonSailor, 2017-09-21:
Yes, I ran the jetson_ros_install.sh script on a TX-1 freshly flashed with Jetpack 3.1.

What other details would you like? 

Attached is the terminal output from the script. Note: this was from the 2nd time running it. The first time the error occurred, I tried logging out, then logged back in and ran it again.

[jetson_ros_install_log.txt](https://github.com/NVIDIA-Jetson/redtail/files/1322165/jetson_ros_install_log.txt)

Alexey-Kamenev, 2017-09-21:
Thank you, that's all we need. It looks like the problem appeared when the new release, 0.21, of MAVROS made into master few days ago, this is the relevant [commit](https://github.com/mavlink/mavros/commit/599c5887ad2aadc124d7e1354f3a1597402118da). I'm testing the fix now but if you don't want to wait for the fix, you can uninstall MAVROS, get the 0.19 from the GitHub and re-build the components.

TritonSailor, 2017-09-22:
The fix solved this issue for me. The jetson_ros_install.sh script completed without error.

PR #6 (2017-09-22): MAVROS 0.21 fix
1. Updated the controller and install script to work with MAVROS 0.21
2. Added .dtb file and instructions to fix TX1 UART issue.

This fixes #5 .
closed 2017-09-22 by TritonSailor