NVIDIA-AI-IOT/redtail #8: Some more details on modified yolo

MarcGyongyosi, 2017-09-29:
Hi team, 

First of all thanks a lot for releasing the code. Great work!

Could you point me to some references on your workflow from training yolo with darknet then converting the cfg and weights to Caffe and making the modifications to yolo? Which script(s) did you use to convert and what is the performance difference between your yolo with TensorRT vs the darknet yolo/yolo2 implementations on a TX1/2? I was only able to find some timing measurements in you GTC slides, but no comparison in terms of mAP or any other detection quality measure. 

Thanks!
Marc

Alexey-Kamenev, 2017-09-30:
Hi Marc,
The way we trained our version of the model was very similar to the original YOLO v1 experiments:
1. Train slightly modified version of YOLO in darknet. The reason we had to change the topology was TensorRT did not support some of the features (like Leaky ReLUs or untied convolutions) at that time. I've added the darknet config file [here](../tree/alexeyk/yolo/models/nets/YOLO).
2. Convert the resulting weights file to Caffe format. The script that we wrote was somewhat hacky :) so we decided not publish it until we make it pretty. However, there are darknet -> Caffe converters like [this](https://github.com/xingwangsfu/caffe-yolo) which hopefully should work.

MarcGyongyosi, 2017-10-01:
Hi Alexey, 

Thanks a lot! This was very helpful and I think this should work for what we want to test. 

If you have any performance metrics to compare TensorRT vs the darknet implementation that would also be very interesting to see. We will run our own benchmark tests anyways, though, so no problem if you can't release them. 

Thanks! 
marc

MarcGyongyosi, 2017-10-19:
Hi Alexey, 

Would you have the original darknet weights file available for reference?

Thanks!
Marc

Alexey-Kamenev, 2017-10-20:
Hi Marc,
I've uploaded the weights to my branch (`alexeyk/yolo`), please do `git pull`. We will merge it into master in the next release along with more YOLO-related info.