NVIDIA-AI-IOT/redtail #12: Simulation on TX1

xuming-shanghai, 2017-10-16:
According to the instructions with simulation, I found most packages(TensorRT-2.1.2.x86_64.cuda-8.0-16-04-tar.bz2, nvidia-docker.......) work on x64 platform, which means it only works on PC not Jetson.
How can I make simulation work on TX1?

Alexey-Kamenev, 2017-10-16:
Running docker on Jetson is not very straightforward, especially things like nvidia-docker etc so we don't support such scenario. The main purpose of Redtail Docker image is to enable development and debugging of the system on the host, not the device. On the device you can just use standard JetPack with our [setup script](../blob/master/ros/scripts/) which installs all required dependencies. You can then run some components on the device, while others (e.g. Gazebo etc) on the host as all of the communication is done via ROS. For example, you can run DNN node on the device, while the controller - in the simulator Docker on the host and so on.

Alexey-Kamenev, 2017-10-17:
I'm closing the issue, feel free to re-open or create a new one.

closed 2017-10-17 by Alexey-Kamenev

xuming-shanghai, 2017-10-18:
Thanks, Alexey.  It's good to me.