NVIDIA-AI-IOT/redtail #14: Does joystick must have?

xuming-shanghai, 2017-10-19:
Hi Guys, 
In wiki page, part : "Flying a drone" --- "Flying"
The article mentioned control the drone using joystick:
left stick:
      up/down: change altitude higher/lower
left/right: yaw
right stick:
      up/down: fly forward/backward
left/right: strafe left/right

Can we use the 8 channels transmitter to replace joystick, 
[http://download.ardupilot.org/epub/_images/FRSkyTaranis.jpg](url)

we can also control roll/pitch/yaw/throttle through transmitter. and control transmitter's switches into different modes(POS HOLD, STABILIZED ....), even into channel 7: kill switch channel 8: offboard switch.

current control data direction:
joystick--->tx1---->pixhawk
Does ROS on tx1 grab joystick's data（analysis, take action...） or just transparent forward to pixhawk through mavlink?

thanks ahead!

Alexey-Kamenev, 2017-10-20:
Hello,
The joystick is currently required as it provides somewhat simplified navigation functionality such as moving forward and turning using one stick (right). You can use left stick to change altitude or rotate. Joystick's buttons are used to enable/disable some of the functionality like DNN navigation (buttons `A`/`B`). The joystick runs in parallel with transmitter. The transmitter itself is used for initial takeoff and switching to `OFFBOARD` mode. Transmitter also has a kill switch which allows to disable vehicle motors in case of emergency. 

As for implementation: ROS `joy` node publishes joystick messages which are processed by our `px4_controller`. The commands from the joystick are fused with our control commands, with joystick always takes precedence over DNN commands so you can override/correct DNN behavior if needed. The controller then publishes resulting command (pose) via MAVROS.
Refer to `px4_controller` code for more information. 

There is a minor inconvenience in such setup as you need at least 2 people: one with transmitter and another with joystick. We found that such setup provides sufficient level of safety during our experiments.

xuming-shanghai, 2017-10-20:
Hi Alexey,
Thanks for the quick response. 
It looks joy stick is a auxiliary device once DNN control the attitude is not expected.
so, my understanding is it's pixhawk its own design logical, if enter offboard mode, it cannot
accept roll/pitch/yaw/ pwm from transmitter to correct/override DNN behavior. That need user purchase one more device.

Alexey-Kamenev, 2017-10-25:
That's correct - once the offboard mode enabled, Pixhawk will ignore movement commands from the transmitter. However, Kill switch as well as Offboard switch will always work regardless of offboard mode. This provides a good safety mechanism: the person with transmitter can always take control back by turning off Offboard mode or using Kill switch if needed. You can probably change the controller code to listen to commands from the transmitter (RC channels) but we haven't explored this.

xuming-shanghai, 2017-10-30:
Thanks, Alexey. I will close this topic.

closed 2017-10-30 by xuming-shanghai