NVIDIA-AI-IOT/redtail #15: Library confusion when compiling OpenCV

cosmicog, 2017-10-20:
Hi,

On wiki, under Test Simulation, without showing correct `libcuda.so`, it gives this error when building;

```
/usr/local/nvidia/lib/libcuda.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
```

Please review my changes and merge them in, from;
[https://github.com/cosmicog/redtail-wiki/commit/f1adf5a3a327f7bcd759d77a5f6ba5657afa3da9](https://github.com/cosmicog/redtail-wiki/commit/f1adf5a3a327f7bcd759d77a5f6ba5657afa3da9)

Alexey-Kamenev, 2017-10-23:
Hello,
Thank you, we haven't seen such problem on our machines or in Docker. What is the system that you are building the code on?

cosmicog, 2017-10-24:
Thank you, My **Nvidia Driver Version: 375.66**, on **Ubuntu 16.04**. I'm not using latest driver because it doesn't work on my machine well. 

With using `-D CUDA_CUDA_LIBRARY=/usr/local/cuda/lib64/stubs/libcuda.so`, I can compile it, also tested with `caffe_ros` node and it is working. Just, I can't feel that the copter on Gazebo is following the road, It draws strange paths. Is it normal?

Alexey-Kamenev, 2017-10-24:
I've added a note to our wiki, thanks. I've found a few similar issues, like [this](https://github.com/opencv/opencv/issues/6577). 
In general, if drivers are properly installed then you should have `libcuda` at this path `/usr/lib/x86_64-linux-gnu/` on your Ubuntu host. When using Docker, `nvidia-docker` will map NVIDIA driver libraries, including `libcuda`, to `/usr/local/nvidia/lib64/` inside the Docker container. 

As for the videos, I would start with checking that everything works in the most basic scenarios. For example, try feeding the same image using `image_pub` node and check the results. There are few test images that you can try which should produce stable results. For example, using [this](../blob/master/ros/packages/caffe_ros/tests/data/rot_c.jpg) image should consistently produce commands that will make the drone fly straight. You can also check the output of DNN node to see what exactly DNN produces.

cosmicog, 2017-10-25:
Thank you for the library information, also for basic scenarios. This issue can be closed for now, but in case if you think that it can be related to similar ones too, I'm leaving it open.

Alexey-Kamenev, 2017-10-25:
I'm going to add more debugging info to our wiki.
Thanks.

closed 2017-10-25 by Alexey-Kamenev