NVIDIA-AI-IOT/redtail #16: Where can I buy the parts for modfy the optical flow?

freaad, 2017-10-22:
Hello.
I am interested in your project. I read the wiki and I am just wondering where can I buy the 6mm lens parts?
I hope that in the wiki where can we buy the extra stuffs not only jetson and iris+.

freaad, 2017-10-30:
please give some tip

nsmoly, 2017-10-31:
Hello, you can use lenses with 6mm focal length, with S-mount, with IR filter and that are good for 1/3" image sensors if you want to replicate our wider angle PX4FLOW setup. For example, [this lens should be good](https://www.lensation.de/product/BM6020ND/). FYI, PX4FLOW uses this [imaging sensor](http://www.onsemi.com/pub/Collateral/MT9V034-D.PDF).
Please make sure that you follow our [instructions](https://github.com/NVIDIA-Jetson/redtail/wiki/3DR-Iris-Setup#px4flow) on re-building PX4FLOW firmware if you change the default focal length for PX4FLOW to 6mm (by default it is 16mm).

freaad, 2017-11-01:
Thank you I will contact that company

Alexey-Kamenev, 2017-11-01:
I'm closing the issue. Feel free to re-open or create a new one.

closed 2017-11-01 by Alexey-Kamenev

Oceasdeep, 2018-06-24:
thanks i see this issue.I want to know ，If I change to a 6 mm lens, do I need to modify other parameters besides changing the focal length parameters?

nsmoly, 2018-06-24:
no, you don't need to modify other parameters (other than focal length in the firmware).

Oceasdeep, 2018-06-27:
@nsmoly thanks a lot. I tried it, and it worked.
I would like to ask another question. How do I control the speed of UAV flying from one coordinate point to another? To achieve autonomous flight, the speed of UAV flight must be adjusted according to different situations during flight. I'm using pixhawk

nsmoly, 2018-08-29:
sorry for delay --  you can set waypoints closer or farther out and control speed to some degree this way. Pixhawk will not exceed max set speed limit, but will try to reach a point as fast as possible and as smooth as possible. So the distance to a waypoint affects your speed. But this is not a precise method. There are also way to do more direct speed control in Pixhawk via MavLink messages.

ashishimpace, 2020-02-28:
@Oceasdeep  As you have changed lens to 6mm, what changes have you observed as compared to default solution ?