NVIDIA-AI-IOT/redtail #17: No movement

griz1112, 2017-10-31:
Set up RedTail. Everything looks like its working but I get no movement on the wheel motors. Channel 1 and channel 3 in r/c out remain at 1500. Numbers are coming from px4_controller output. I see commands being sent and responded to by mavros. Any ideas where to look? How is it supposed to behave when you first start it up?

Alexey-Kamenev, 2017-11-01:
Just to make sure, what platform are you using? Is it APM or you changed the code to work with PX4?
What happens if you change RC channel values via MAVROS directly (i.e. `rosservice call`)?

griz1112, 2017-11-02:
APM. What I see in QGC is the mode keeps switching between manual and unknown. I will try the rosservice call and let you know. Do  you start up px4_controller the same way as for the iris? Unless I change the mode under the drone definition it tries to use offboard which won't work with APM. So do I need to specify its apm rover somewhere in the command line that starts it?

griz1112, 2017-11-02:
There is no mavros/rc/override that shows up with rosservice list -n. However if I run rqt its there.

Alexey-Kamenev, 2017-11-02:
`px4_controller` currently supports 2 vehicle types: `drone` and `apmrover` via `vehicle_type` argument. The default is `drone` so it likely explains why you haven't seen any movements in your rover. Also, sorry for confusion re: the service - RC override plugin (`rc_io`) is not a service, it is topic (I pasted the command from `param` plugin which **is** a service). So the command line should look something like that:
```
rosrun px4_controller px4_controller_node _vehicle_type:=apmrover
```
with other necessary parameters.

Before running the controller, you can verify that everything else (MAVROS, connections etc) are working as expected. First, check what PWM is at servo outputs:
```
rostopic echo /mavros/rc/out
``` 
Move the stick(s) - the values should change.

Second, try controlling the motors with MAVROS:
```
rostopic pub /mavros/rc/override mavros_msgs/OverrideRCIn '{channels: [65535, 1300, 65535, 65535, 65535, 65535, 65535, 65535, 65535]}'
```
Note that you may need to set `SYSID_MYGCS` parameter first to allow RC override from the code.

griz1112, 2017-11-02:
Thanks that part is working now but getting a fatal error in the takeoff part. 
[FATAL] [1509649000.673626672]: ASSERTION FAILED
	file = /home/nvidia/ws/src/px4_controller/src/px4_controller.cpp
	line = 95
	cond = is_initialized_

Joystick is hooked up and working via rostopic. FC is armed and in the right mode.

Alexey-Kamenev, 2017-11-02:
Looks like a merge issue. Can you please try `alexeyk/rover` branch and see if it works?

griz1112, 2017-11-03:
Reflashed and installed the rover git. Same results. If I set the mode and disarm in QGC. Then run px4_controller I see it arm and change to the proper mode. However the tone from buzzer indicating its armed happens just after it throws that error. Wondering if its not waiting long enough and missing the message from mavros that its armed so initialized.

griz1112, 2017-11-03:
rostopic pub /mavros/rc/override mavros_msgs/OverrideRCIn '{channels: [65535, 1300, 65535, 65535, 65535, 65535, 65535, 65535, 65535]}'
ERROR: Field [channels] has incorrect number of elements: 9 != 8

Alexey-Kamenev, 2017-11-03:
Re: number of elements: remove the last number from the list.

Just to make sure, you are getting the same assert error message? You don't need to re-flash, just rebuild controller code from my branch. Something like this:
1. SSH to Jetson.
2.
```
cd ~/redtail
git pull
git checkout alexeyk/rover
cd ~/ws
catkin_make px4_controller px4_controller_node
```

griz1112, 2017-11-03:
Managed to get past that error by building mavros from source. I've seen this before trying to use ROS with APM. Still no motion of the wheels using dnn or rostopic. I'll have to look into that tomorrow.

griz1112, 2017-11-03:
Should I be seeing messages to /mavros/setpoint_position/local published? Also what version of apm firmware did you test with?

Alexey-Kamenev, 2017-11-03:
No, `apmrover` mode uses simple RC override so it does not use/publish pose. First need to understand why `/mavros/rc/override` is not working via `rostopic pub`. Unfortunately, without access to the vehicle it's hard to tell what's going on but try looking at other RC topics like `rc/in` and `rc/out`. 
As for the version - we use whatever APM comes pre-installed with Erle Brain 3 (2.6?)

griz1112, 2017-11-04:
I'm using 3.2 Don't think 2.6 will even run on a pixhawk 2.1. I see mavros/rc/override topic published with numbers. Rc out stays at 1500 for the two channels no matter what. If you move the rover around you see the dnn trying to correct in the output but nothing changes.

griz1112, 2017-11-04:
rosparam set /mavros/system_id 1 fixed it. Thanks for all the help.

Alexey-Kamenev, 2017-12-04:
Not sure if this is still relevant but FYI anyway: I've just [merged](../pull/26) a few changes/samples which make the work with rovers a bit easier. We tested it with a couple of rovers and DNNs that we have here. Note that this is still RC-override based control.
I'm closing the issue now, feel free to re-open or create new one.

closed 2017-12-04 by Alexey-Kamenev