NVIDIA-AI-IOT/redtail #18: Nvidia joystick

xuming-shanghai, 2017-11-02:
Hi guys,
Can I purchase a wireless(bluetooth) Nvidia shield controller? Is this joystick same as yours?
I am not sure the bluetooth connection whether generate a input device node(/dev/input/js0).
Please help confirm.

purchase link:
https://item.taobao.com/item.htm?spm=a230r.1.14.106.34de5260Rhf5F7&id=40828751854&ns=1&abbucket=9#detail

Alexey-Kamenev, 2017-11-02:
That's a good question, I currently only have an older Shield controller (the one that uses WiFi Direct, not Bluetooth). Let me find the new one and get back to you. In general though, as long as it can be recognized by Linux as joystick device (e.g. visible under `/dev/input/js*`) it should work fine. There might be some changes required in case the button/axis mapping is different from what we support now (Shield, XBox Wired and XBox Wireless) but it's a minor change.

xuming-shanghai, 2017-11-03:
Thanks Alexey, that's the key point whether bluetooth will create an input node works under ubuntu linux. 
In China, it's hard to purchase single shield controller without host box. 
Above link, it is the unique vendor on TaoBao(alibaba) I can found.
so, thank you help me try a new one.

Alexey-Kamenev, 2017-11-03:
I've successfully tested the new controller and had to make a small change to the code as the new Shield has slightly different axis mapping. I've [merged](../pull/19) the changes to master branch, please sync. Use `shield_2017` joystick type in `px4_controller` when running with new Shield, something like that:
```
rosrun px4_controller px4_controller_node _joy_type:=shield_2017
```
Bluetooth connection worked fine on my Ubuntu 16.10. Make sure to enable Bluetooth and pair joystick, device name should be something like `NVIDIA Controller v01.04` (your version number might differ).

xuming-shanghai, 2017-11-06:
Thanks Alexey. That's great.

Alexey-Kamenev, 2017-11-06:
Closing the issue. Feel free to re-open or create new one.

closed 2017-11-06 by Alexey-Kamenev