NVIDIA-AI-IOT/redtail #20: QGround connect to pixhawk failure with  MB12xx sonar enabled

xuming-shanghai, 2017-11-06:
Hi Alexey
Do you ever encounter this issue:
set "SENS_EN_MB12xx" enabled and close Qground, reboot pixhawk, then wait few seconds open Qground again, pixhawk never connected to qground successfully.
To double check this issue, I erase pixhawk flash, program binary again, set this parameter several times.
This issue not related whether px4flow i2c connected to pixhawk or not.
**_Program online standard version(stable) not found this issue._**
can you help check this? Thanks, otherwise I need change firmware version if I want to MB12xx sonar, not Lidar.

firmware version:  PX4 firmware v1.4.4.
QGroundControl version: 3.2.4/2.9.5......

xuming-shanghai, 2017-11-08:
Hi Alexey
Maybe I made a mistake. Pixhawk led turn green when px4flow on with FW version: 1.4.4.
It's no necessary to set "SENS_EN_MB12xx" enabled when px4flow equipped on pixhawk.
Although px4flow has sonar MB12xx and MT9v034, but this is the integrated combination module; To pixhawk, px4flow is one sensor. 
So, if using independent MB12xx I2C interface sensor, I guess this parameter maybe needed.

Alexey-Kamenev, 2017-11-09:
That's correct, you don't need to enable sonar when using it as a part of PXFLOW sensor. In our wiki we only mention the [parameters](../wiki/3DR-Iris-Setup#pixhawk-parameters) that you need to set but we probably should mention this case specifically to avoid confusion.

xuming-shanghai, 2017-11-10:
Yeah, that maybe 1.4.4 version bug with MB12xx sensor enabled, but looks fixed in recent release.
Anyway, it's pixhawk bug, we may ignore it and close this topic.
Thanks, Alexey.

closed 2017-11-10 by xuming-shanghai