NVIDIA-AI-IOT/redtail #21: Fake camera stream and use local video file instead

xuming-shanghai, 2017-11-08:
Hi Alexey and guys:
Can you help provide the first view video recorded by your drone when your were testing in forest?
It's hard to simulate forest environment in shanghai city.
so, I assume let my drone read your video file(not  my camera), then drone thoughts it is in forest.
now I put my drone in outdoor playground, other steps are same as yours except input video steam are faked.
If observed the drone's flight route is along with the road direction in the video, so I think I copied your demo.
Is it feasible about my thought?
Thanks ahead!

Alexey-Kamenev, 2017-11-09:
Hello,
That sounds like a good plan. We haven't uploaded the raw drone FPV video, let me check if that can be done quickly.
For now, you can first try playing our [test images](../tree/master/ros/packages/caffe_ros/tests/data) using [image_pub](../tree/master/ros/packages/image_pub) node to make sure your build produces expected results on most basic scenarios. Essentially you need to make sure that if you play, say, "right" image continuously then your drone makes something that vaguely resembles a circle.
Next, you can feed videos from [IDSIA Forest Trails](http://people.idsia.ch/~guzzi/DataSet.html) dataset using the same `image_pub` node. 
I would recommend doing both steps in simulator first before going outside.
Let us know how it works and don't forget to record and share the videos! :)

xuming-shanghai, 2017-11-10:
Let me ensure the drone can position hold with px4low first, today I find a wide and open place to test optical flow. Unfortunately, the drone always drifts from right to left.  But when I hand it with 1 meter distance to ground, the led can turn green.
I have few experience on px4flow, could you help me analysis what happened according your experience. Some parameters are set incorrectly?
btw, I have calibrate the sensor with 1m distance.
here is the parameters I referred:

Name         Default         Access         Comment BFLOW_F_THLD         5000         RW         This parameter is a feature threashold and limits the quality of patterns that are used to calculate the bottom flow. For low values (e.g. 10) almost every pattern is taken, for higher values (e.g. 100) only significant patters are taken. BFLOW_V_THLD         30         RW         This is a pattern correlation threashold for filtering bad matches. Lower means only strong correlations are accepted. BFLOW_HIST_FIL         0         RW         1: Flow histogram filter is ON, 0: OFF BFLOW_GYRO_COM         1         RW         1: Gyro compensation is ON, 0: OFF BFLOW_LP_FIL         0         RW         1: Lowpass filter on flow output is ON, 0: OFF BFLOW_W_NEW         0.3         RW         Flow lowpass filter gain DEBUG         1         RW         1: Debug messages ON, 0: OFF GYRO_SENS_DPS         250         RW         Gyroscope sensitivity: 250, 500, 2000 (dps) GYRO_COMP_THR         0.01         RW         Gyro compensation threshold (dps): Gyro data lower than this threshold is not compensated to prevent drift IMAGE_WIDTH         64         R         Image width (pixels) IMAGE_HEIGHT         64         R         Image height (pixels) IMAGE_L_LIGHT         0         RW         1: Image sensor low light mode ON, 0: OFF IMAGE_NOISE_C         1         RW         1: Image sensor noise correction ON, 0: OFF IMAGE_TEST_PAT         0         RW         1: Gray-shaded test pattern mode ON, 0: OFF LENS_FOCAL_LEN         16         RW         Focal length of lens (mm) POSITION         0         RW         0: Only position 0 is used (Bottom: 0, Front: 1, Top: 2, Back: 3, Right: 4, Left: 5) SONAR_FILTERED         0         RW         1: Kalman filter on sonar output is ON, 0: OFF SONAR_KAL_L1         0.8461         RW         Sonar Kalman gain L1 SONAR_KAL_L2         6.2034         RW         Sonar Kalman gain L2 SYS_ID         81         RW        MAVLink System ID SYS_COMP_ID         50         RW        MAVLink Component ID SYS_SENSOR_ID         77         RW        MAVLink Sensor ID SYS_TYPE         0         RW        MAVLink System Type SYS_AP_TYPE         1         RW        MAVLink Autopilot Type SYS_SW_VER         13XX         R         Software Version SYS_SEND_STATE         1         RW         1: Send MAVLink Heartbeat, 0: Not USART_2_BAUD         115200         R         Baudrate USART 2 USART_3_BAUD         115200         R         Baudrate USART 3 (Data Output) USB_SEND_VIDEO         1         RW         1: Send video over USB, 0: Not USB_SEND_FLOW         1         RW         1: Send flow over USB, 0: Not USB_SEND_GYRO         1         RW         1: Send gyro data over USB, 0: Not USB_SEND_FWD         0         RW         1: Send forwarded flow over USB, 0: Not USB_SEND_DEBUG         1         RW         1: Send debug msgs over USB, 0: Not VIDEO_RATE         150         RW         Time in milliseconds between images of video transmission VIDEO_ONLY         0         RW         1: High resolution video mode is ON, 0: OFF | Name         Default         Access         Comment BFLOW_F_THLD         5000         RW         This parameter is a feature threashold and limits the quality of patterns that are used to calculate the bottom flow. For low values (e.g. 10) almost every pattern is taken, for higher values (e.g. 100) only significant patters are taken. BFLOW_V_THLD         30         RW         This is a pattern correlation threashold for filtering bad matches. Lower means only strong correlations are accepted. BFLOW_HIST_FIL         0         RW         1: Flow histogram filter is ON, 0: OFF BFLOW_GYRO_COM         1         RW         1: Gyro compensation is ON, 0: OFF BFLOW_LP_FIL         0         RW         1: Lowpass filter on flow output is ON, 0: OFF BFLOW_W_NEW         0.3         RW         Flow lowpass filter gain DEBUG         1         RW         1: Debug messages ON, 0: OFF GYRO_SENS_DPS         250         RW         Gyroscope sensitivity: 250, 500, 2000 (dps) GYRO_COMP_THR         0.01         RW         Gyro compensation threshold (dps): Gyro data lower than this threshold is not compensated to prevent drift IMAGE_WIDTH         64         R         Image width (pixels) IMAGE_HEIGHT         64         R         Image height (pixels) IMAGE_L_LIGHT         0         RW         1: Image sensor low light mode ON, 0: OFF IMAGE_NOISE_C         1         RW         1: Image sensor noise correction ON, 0: OFF IMAGE_TEST_PAT         0         RW         1: Gray-shaded test pattern mode ON, 0: OFF LENS_FOCAL_LEN         16         RW         Focal length of lens (mm) POSITION         0         RW         0: Only position 0 is used (Bottom: 0, Front: 1, Top: 2, Back: 3, Right: 4, Left: 5) SONAR_FILTERED         0         RW         1: Kalman filter on sonar output is ON, 0: OFF SONAR_KAL_L1         0.8461         RW         Sonar Kalman gain L1 SONAR_KAL_L2         6.2034         RW         Sonar Kalman gain L2 SYS_ID       [... cut ...] Not USB_SEND_FLOW         1         RW         1: Send flow over USB, 0: Not USB_SEND_GYRO         1         RW         1: Send gyro data over USB, 0: Not USB_SEND_FWD         0         RW         1: Send forwarded flow over USB, 0: Not USB_SEND_DEBUG         1         RW         1: Send debug msgs over USB, 0: Not VIDEO_RATE         150         RW         Time in milliseconds between images of video transmission VIDEO_ONLY         0         RW         1: High resolution video mode is ON, 0: OFF

Alexey-Kamenev, 2017-11-13:
Making PX4FLOW can be a bit tricky sometimes indeed. Please make sure that you read and followed our [PX4FLOW instructions](https://github.com/NVIDIA-Jetson/redtail/wiki/3DR-Iris-Setup#px4flow). That is, check the firmware, lenses and camera video quality. If everything is setup correctly then PX4FLOW should initialize when you lift the drone above the ground about 20-30 cm (as described [here](https://github.com/NVIDIA-Jetson/redtail/wiki/Launch-Sequence-and-Flying#powering-on)).
It might be also useful to see the output of PX4FLOW module in QGroundControl: connect PX4FLOW to your computer USB, run QGC and select `Widgets -> Analyze` menu. Add PX4FlOW topics (I don't remember the names but it should be clear from topic names) and check that x,y coordinates are stable.
If PX4FLOW is configured properly and there is enough texture on the ground then the drone should not drift while flying.

P.S. I'm re-assigning the issue to Nikolai as I'm at the conference now.

xuming-shanghai, 2017-11-15:
Thanks, I will try.
btw, image_pub node looks not built in ws.
Can you make changes as below, it works well on my tx1.

diff --git a/ros/scripts/jetson_ros_install.sh b/ros/scripts/jetson_ros_install.sh
index 1d036bc..e0c207d 100755
--- a/ros/scripts/jetson_ros_install.sh
+++ b/ros/scripts/jetson_ros_install.sh
@@ -143,6 +143,7 @@ if [ ! -L "$CATKIN_WS/src/caffe_ros" ]; then
     # Create symlinks to catkin workspace.
     ln -s $HOME/redtail/ros/packages/caffe_ros $CATKIN_WS/src/
     ln -s $HOME/redtail/ros/packages/px4_controller $CATKIN_WS/src/
+    ln -s $HOME/redtail/ros/packages/image_pub $CATKIN_WS/src/
 fi

 echo "Installing dependencies..."
@@ -162,6 +163,8 @@ echo "Building caffe_ros package..."
 catkin_make caffe_ros_node
 echo "Building px4_controller package..."
 catkin_make px4_controller_node
+echo "Building image_pub package..."
+catkin_make image_pub_node

 # Environment setup.
 echo "source $CATKIN_WS/devel/setup.bash" >> $HOME/.bashrc

diff --git a/ros/packages/image_pub/CMakeLists.txt b/ros/packages/image_pub/CMakeLists.txt
index a2f31ff..05c3704 100644
--- a/ros/packages/image_pub/CMakeLists.txt
+++ b/ros/packages/image_pub/CMakeLists.txt
@@ -6,6 +6,14 @@ find_package(catkin REQUIRED COMPONENTS
   std_msgs
 )

+catkin_package(
+# INCLUDE_DIRS include
+  LIBRARIES image_pub_node
+  CATKIN_DEPENDS roscpp std_msgs
+# DEPENDS system_lib
+)
+
+
 ###########
 ## Build ##
 ###########

xuming-shanghai, 2017-11-20:
Hi Alexey, Nikolai 
My px4flow is not ready now.
So, I use gazebo simulation on x64 system to simulation image pub and  offboard flight control.
The configuration is a bit different with yours, my configuration is real tx1(image pub, mavros, trailnet_dnn, yolo_dnn, px4-controller) + x64 PC(gazebo, qground), that can workaround your nvidia graphics card on PC and also without installing docker.io. I think the principle is the same.
As you mentioned before, when pub a jpg image(redtail/ros/packages/caffe_ros/tests/data/rot_c.jpg) with repeatable mode, the drone circle and circle....
I think it's the reasonable logic, the same frame/image input, the same angle output; so, circle....., but it's not the expected behavior, the previous one frame was same as current one, the dnn network should ignore it, and keep the drone stop and stay in current position.  Another test, when I ctrl+c image pub node, there was no image data input, the drone stopped and  stay in current position, it looks safe to us.

Alexey-Kamenev, 2017-11-21:
Your setup is another good option to test/develop the components. Basically, since all our components are implemented as ROS nodes, you can mix and match workstations, Jetsons and Dockers in any way you like :) We decided to provide and support just 2 scenarios, one based on real Jetson and another - Docker. Such scenarios are relatively easily reproducible and repeatable (especially with Docker).

As for the `image_pub` node and what to do when the same image is published repeatedly: you can change the code to check for image similarity and do something else (like stopping) in case your metric reaches certain threshold. There is currently no safety mechanisms other than very basic ones like when controller dies, PX4 turns off the offboard mode and switches back to position hold.

nsmoly, 2017-11-22:
@xuming-shanghai: just to explain what's going on if the focal length is wrong - PX4FLOW cannot estimate velocity properly (due to wrong camera model) and so this leads to overcontrol in Pixhawk. It will stabilize, but it will "float" around (typically side to side)

ktnguyen2, 2019-07-15:
@xuming-shanghai Did you use NVIDIA GPU on your PC? If not, how did you work around that?