NVIDIA-AI-IOT/redtail #22: Please include labels_map.txt

mscribe, 2017-11-13:
I ran [idsia_trails_dataset_digits.py](https://github.com/NVIDIA-Jetson/redtail/blob/master/models/dataset/idsia_trails_dataset_digits.py), which created train_map.txt and val_map.txt.  How can I get my hands on labels_map.txt as seen in the image found [here](https://github.com/NVIDIA-Jetson/redtail/wiki/Datasets)?

[image]

mscribe, 2017-11-13:
I noticed inside the CIFAR-10 demo, that labels.txt was simply a list of human readable text labels that mapped to numeric values in train.txt in a zero-based manner.  So, I created the following labels_map.txt for ForestTrails with the following contents:
```
lc
sc
rc
```
I also squashed the images to 320x180 when creating the dataset, based on what I saw in the model training instruction screenshots.

Alexey-Kamenev, 2017-11-13:
Yes, you are correct, it's just a simple text file that maps label indices to names. Pushed with 37ba895.

mscribe, 2017-11-14:
Thanks for the quick response!

closed 2017-11-14 by mscribe