NVIDIA-AI-IOT/redtail #23: Some mistakes in everything.launch?

xuming-shanghai, 2017-11-15:
In file everything.lunch, there is one line:
 **<arg name="object_model_path"    default="/home/nvidia/redtail/models/pretrained/yolo-relu.caffemodel" />** 

but not found this file in models/pretrained/ directory:
**README.md	Initial commit	2 months ago
TrailNet_SResNet-18.caffemodel	Initial commit	2 months ago
TrailNet_SResNet-18.prototxt	Initial commit	2 months ago
yolo-relu.caffemodel.00	Initial commit	2 months ago
yolo-relu.caffemodel.01	Initial commit	2 months ago
yolo-relu.caffemodel.02	Initial commit	2 months ago
yolo-relu.prototxt	Initial commit	2** 

Now If changing "yolo-relu.caffemodel" to "yolo-relu.caffemodel.00" "yolo-relu.caffemodel.01" "yolo-relu.caffemodel.02", any one file in these 3 file, issue encountered:

**[FATAL] [1510726990.784716595]: Failed to parse network: /home/nvidia/redtail/models/pretrained/yolo-relu.prototxt, /home/nvidia/redtail/models/pretrained/yolo-relu.caffemodel.00**

Now run yolo-dnn node independently, 
rosrun caffe_ros caffe_ros_node __name:=object_dnn _prototxt_path:=/home/nvidia/redtail/models/pretrained/yolo-relu.prototxt _model_path:=/home/nvidia/redtail/models/pretrained/yolo-relu.caffemodel.02 utput_layer:=fc25 _inp_scale:=0.00390625 _inp_fmt:="RGB" _post_proc:="YOLO" _obj_det_threshold:=0.2 _use_fp16:=true

same issue happened:
[FATAL] [1510727562.919367634]: Failed to parse network: /home/nvidia/redtail/models/pretrained/yolo-relu.prototxt, /home/nvidia/redtail/models/pretrained/yolo-relu.caffemodel.02
Segmentation fault

Alexey-Kamenev, 2017-11-15:
Please follow the instructions for YOLO model [here](../tree/master/models/pretrained).

xuming-shanghai, 2017-11-15:
oh...oh....
Thank you.

closed 2017-11-15 by xuming-shanghai