NVIDIA-AI-IOT/redtail #25: USB Patch

eanmikale, 2017-12-03:
Hi, 
Foremost, thanks for releasing this project. We have questions concerning the USB Patch for Jetpack 3.1. In particular, we are interested in determining where the .dtb files should be found. We have searched in the 64_TX2 directory, and found .dtb files that were already located in the following directories Linux_for_Tegra_tx2/bootloader, Linux_for_Tegra_tx2/kernel/dtb, and Linux_for_Tegra_tx2/rootfs/boot. Also, does this patch address the same issue as the JetsonHacks/ACMModule fix? Either fix, we hope will prevent us from having to build a kernel to access our teensy microcontroller. Thank you for your time.

Alexey-Kamenev, 2017-12-04:
Hello,
You should overwrite your local files with the ones from our repo by following [the instructions](../tree/master/tools/install/j120-tx2-patch). You may backup original files if you like.
I'm not aware of `JetsonHacks/ACMModule` fix so I cannot comment on that. One way to compare the patches would be to decompile dtb files into source text (dts) using `dtc` tool and compare the results.

Even better, if you are using J-120 from Auvidea then you can now take patches from [Auvidea website](https://auvidea.com/firmware/).

Alexey-Kamenev, 2017-12-11:
Closing the issue, feel free to re-open or create a new one.

closed 2017-12-11 by Alexey-Kamenev