NVIDIA-AI-IOT/redtail #29: Webcam model

alessandro-montanari, 2017-12-12:
What webcam did you use and what fps where you able to achieve with it?
Thanks for opening up this work!

Alexey-Kamenev, 2017-12-12:
We tried few cameras but the one you see on the videos is Microsoft LifeCam HD-5000 web camera. Other cameras should work too as long as they work with V4L (one easy way to check is to run one of our GStreamer pipelines using GStreamer or `gscam` node).
The camera was running at 30 fps, our TrailNet was running at the same rate while YOLO - at 1 fps. All rates are configurable via node parameters and/or launch files.
While our usage of cheap web camera was intentional, you don't have to do that: our system works just fine with good cameras like ZED.

alessandro-montanari, 2017-12-12:
Thanks for the info.
Actually I am planning to use a cheap webcam as well because it will be mounted on a drone and it is more cost effective in case of crash :)

Alexey-Kamenev, 2017-12-13:
Closing the issue, feel free to re-open or create a new one.

closed 2017-12-13 by Alexey-Kamenev