NVIDIA-AI-IOT/redtail #30: Build issues on x86_64

TritonSailor, 2017-12-12:
I would like to run redtail on my GTX1070 laptop for controlling a small FPV vehicle remotely.

While trying to build redtail on x86_64, I'm getting cmake errors related to Qt5 such as the example below.  I already have the Qt5 packages on my system.  What am I missing here?  What other info can I provide to help solve this issue? install log and dpkg -l output are attached

CMake Error at /usr/share/cmake-3.5/Modules/FindCUDA.cmake:1693 (add_executable):
  Target "caffe_ros_node" links to target "Qt5::Core" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  caffe_ros/CMakeLists.txt:124 (cuda_add_executable)

[log_121217.txt](https://github.com/NVIDIA-Jetson/redtail/files/1552277/log_121217.txt)

[dpkgList.txt](https://github.com/NVIDIA-Jetson/redtail/files/1552297/dpkgList.txt)

Alexey-Kamenev, 2017-12-12:
It seems like you are running our Jetson install script (`jetson_ros_install.sh`) on your x64 machine directly. We have not tested this scenario and it's not supported - the script is intended to be used only with Jetson/JetPack. I can suggest 2 options:
1. Use our [Docker image](../wiki/Testing-in-Simulator#redtail-docker) which has all the components required to run redtail. Once you build and run the container, you can set up the ROS nodes in any way you like, for example, run the controller node on your vehicle, while DNNs - on your laptop (from Docker).
2. Try fixing the script. When you do that, I strongly recommend doing so from a Docker container to avoid polluting your host system. You don't have to use our image, but you can use it as a starting point and skip the components you don't need (e.g. Gazebo etc).

TritonSailor, 2017-12-12:
Thanks you for the suggestions.  Either of those should work for me.  The Docker image solution seems fast and easy, except that it appears to be a 6.21GB file, and unfortunately I'm limited by a slow rural internet connection.

For installing directly on my machine, it would help if you could suggest what dependencies are required to build the redtail ros nodes?  Other than what the script installed, I built opencv2.4.13 from source after the script complained about wrong opencv version.

I tried manually building caffe_ros_node in the workspace setup by jetson_ros_install.sh

I understand from the output that it's a linker error, but I'm confused how to solve it.  I do have qt5 on my machine.  I.E. qgroundcontrol builds from source without any errors.

chris@GL702VSK:~/ws$ catkin_make caffe_ros_node 
Base path: /home/chris/ws
Source space: /home/chris/ws/src
Build space: /home/chris/ws/build
Devel space: /home/chris/ws/devel
Install space: /home/chris/ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/chris/ws/build"
####
####
#### Running command: "make caffe_ros_node -j8 -l8" in "/home/chris/ws/build"
####
[ 25%] Linking CXX executable /home/chris/ws/devel/lib/caffe_ros/caffe_ros_node
/usr/bin/ld: cannot find -lQt5::Core
/usr/bin/ld: cannot find -lQt5::Gui
/usr/bin/ld: cannot find -lQt5::Widgets
/usr/bin/ld: cannot find -lQt5::Test
/usr/bin/ld: cannot find -lQt5::Concurrent
/usr/bin/ld: cannot find -lQt5::OpenGL
collect2: error: ld returned 1 exit status
caffe_ros/CMakeFiles/caffe_ros_node.dir/build.make:203: recipe for target '/home/chris/ws/devel/lib/caffe_ros/caffe_ros_node' failed
make[3]: *** [/home/chris/ws/devel/lib/caffe_ros/caffe_ros_node] Error 1
CMakeFiles/Makefile2:897: recipe for target 'caffe_ros/CMakeFiles/caffe_ros_node.dir/all' failed
make[2]: *** [caffe_ros/CMakeFiles/caffe_ros_node.dir/all] Error 2
CMakeFiles/Makefile2:909: recipe for target 'caffe_ros/CMakeFiles/caffe_ros_node.dir/rule' failed
make[1]: *** [caffe_ros/CMakeFiles/caffe_ros_node.dir/rule] Error 2
Makefile:485: recipe for target 'caffe_ros_node' failed
make: *** [caffe_ros_node] Error 2
Invoking "make caffe_ros_node -j8 -l8" failed

Alexey-Kamenev, 2017-12-12:
My guess would be OpenCV versions issue. `caffe_ros` by itself does not use any `qt` libraries so it is most likely coming from OpenCV. Having multiple versions of OpenCV is a pain (to put it mildly) so this where Docker might help. Another idea would be to build OpenCV without QT (and other redundant in this case components, like samples, tests etc) and see if it helps.
The main reason our code uses version 2.4.13 is that this is the version installed by JetPack so we wanted to avoid building OpenCV in our scripts.

TritonSailor, 2017-12-13:
I was able to build caffe_ros after rebuilding OpenCV 2.4.13 without QT.  The docker image is still downloading...

However, now there is an issue building caffe_ros_tests that is giving me troubles. 

tests.cpp:(.text+0x42): undefined reference to `cv::imread 

Log is pasted below.  

Searching suggested that missing opencv_imgcodecs could be the issue, but it seems to be on my system.
I have verified that opencv_imgcodecs.so is present on my LD_LIBRARY_PATH 

chris@GL702VSK:~/ws$ echo $LD_LIBRARY_PATH
/usr/local/lib:/home/chris/mkdev/opencv/build/lib:/home/chris/ws/devel/lib:/opt/ros/kinetic/lib:/home/chris/Qt/5.9.1/gcc_64/lib:/usr/local/cuda-8.0/lib64:/usr/local/cuda-8.0/targets/x86_64-linux/lib/:/usr/lib/x86_64-linux-gnu/gazebo-7/plugins

chris@GL702VSK:~/ws$ locate opencv_imgcodecs.so
/home/chris/mkdev/opencv/build/lib/libopencv_imgcodecs.so
/home/chris/mkdev/opencv/build/lib/libopencv_imgcodecs.so.3.3
/home/chris/mkdev/opencv/build/lib/libopencv_imgcodecs.so.3.3.0
/usr/local/lib/libopencv_imgcodecs.so
/usr/local/lib/libopencv_imgcodecs.so.3.3
/usr/local/lib/libopencv_imgcodecs.so.3.3.0

chris@GL702VSK:~/ws$ catkin_make caffe_ros_tests 
Base path: /home/chris/ws
Source space: /home/chris/ws/src
Build space: /home/chris/ws/build
Devel space: /home/chris/ws/devel
Install space: /home/chris/ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/chris/ws/build"
####
####
#### Running command: "make caffe_ros_tests -j8 -l8" in "/home/chris/ws/build"
####
[ 25%] Building CXX object gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 50%] Linking CXX shared library libgtest.so
[ 50%] Built target gtest
[ 75%] Building CXX object caffe_ros/CMakeFiles/caffe_ros_tests.dir/tests/tests.cpp.o
[100%] Linking CXX executable /home/chris/ws/devel/lib/caffe_ros/caffe_ros_tests
CMakeFiles/caffe_ros_tests.dir/tests/tests.cpp.o: In function `readImage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
tests.cpp:(.text+0x42): undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
collect2: error: ld returned 1 exit status
caffe_ros/CMakeFiles/caffe_ros_tests.dir/build.make:117: recipe for target '/home/chris/ws/devel/lib/caffe_ros/caffe_ros_tests' failed
make[3]: *** [/home/chris/ws/devel/lib/caffe_ros/caffe_ros_tests] Error 1
CMakeFiles/Makefile2:934: recipe for target 'caffe_ros/CMakeFiles/caffe_ros_tests.dir/all' failed
make[2]: *** [caffe_ros/CMakeFiles/caffe_ros_tests.dir/all] Error 2
CMakeFiles/Makefile2:941: recipe for target 'caffe_ros/CMakeFiles/caffe_ros_tests.dir/rule' failed
make[1]: *** [caffe_ros/CMakeFiles/caffe_ros_tests.dir/rule] Error 2
Makefile:498: recipe for target 'caffe_ros_tests' failed
make: *** [caffe_ros_tests] Error 2
Invoking "make caffe_ros_tests -j8 -l8" failed

Alexey-Kamenev, 2017-12-13:
It looks like it's the wrong version of `libopencv_imgcodecs`: in your output, it's 3.3, not 2.4.

TritonSailor, 2018-02-07:
Awesome, nice work Alexey! I was able to get caffe_ros_node running on my x86_64 laptop by building with your alexeyk/2.0 branch.  Now I've got the tx1 running trailnet onboard the rover, and the yolo node running on my gtx1070 laptop.

closed 2018-02-07 by TritonSailor