NVIDIA-AI-IOT/redtail #33: Build Issues - Cmake Failed

eanmikale, 2018-01-05:
We are using Jetpack 3.1, as well as OpenCV 2.4.13 and are seeing the following build issues. Any assistance would be greatly appreciated. Also, the necessary environment for the Erle Brain 3 was lightly discussed. Is the default environments via Erle Brain docs enough, specifically for the Rover? Thanks again. 

-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - angles
-- ~~  - caffe_ros
-- ~~  - camera_calibration_parsers
-- ~~  - image_transport
-- ~~  - camera_info_manager
-- ~~  - gscam
-- ~~  - px4_controller
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'angles'
-- ==> add_subdirectory(angles)
-- +++ processing catkin package: 'caffe_ros'
-- ==> add_subdirectory(caffe_ros)
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
CMake Error at caffe_ros/CMakeLists.txt:15 (CUDA_CUDART_LIBRARY):
  Unknown CMake command "CUDA_CUDART_LIBRARY".

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
Building px4_controller package...
Base path: /home/nvidia/ws
Source space: /home/nvidia/ws/src
Build space: /home/nvidia/ws/build
Devel space: /home/nvidia/ws/devel
Install space: /home/nvidia/ws/install
####
#### Running command: "cmake /home/nvidia/ws/src -DCATKIN_DEVEL_PREFIX=/home/nvidia/ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/ws/install -G Unix Makefiles" in "/home/nvidia/ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - angles
-- ~~  - caffe_ros
-- ~~  - camera_calibration_parsers
-- ~~  - image_transport
-- ~~  - camera_info_manager
-- ~~  - gscam
-- ~~  - px4_controller
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'angles'
-- ==> add_subdirectory(angles)
-- +++ processing catkin package: 'caffe_ros'
-- ==> add_subdirectory(caffe_ros)
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
CMake Error at caffe_ros/CMakeLists.txt:15 (CUDA_CUDART_LIBRARY):
  Unknown CMake command "CUDA_CUDART_LIBRARY".

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Alexey-Kamenev, 2018-01-08:
The main problem seems to be this:
```
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY)
```

There is something funky with CUDA configuration, can you please check your paths etc? 
Have you selected CUDA Toolkit when installing JetPack as specified [here](../wiki/Jetson-Setup)?
Have you made any changes to CUDA after install? Can you build/run any of CUDA samples?

Alexey-Kamenev, 2018-02-13:
Closing the issue, feel free to re-open or create a new one.

closed 2018-02-13 by Alexey-Kamenev