NVIDIA-AI-IOT/redtail #34: Auvidea J-120 + Jetson TX2 issue: Fixed in JetPack 3.2?

kometa-triatlon, 2018-01-05:
Hi,

[This page](https://github.com/NVIDIA-Jetson/redtail/blob/master/tools/install/j120-tx2-patch/README.md) claims that the USB issue with the J120 board must be fixed in the JetPack release after 3.1. Recently I found the version 3.2 developer preview is here, but I cannot see anything related to that in the [release notes](http://docs.nvidia.com/jetpack-l4t/3.2/index.html#developertools/mobile/jetpack/l4t/3.2/jetpack_l4t_release_notes.htm).

Can you tell me whether the issue was fixed in the version 3.2 or I'd better use 3.1 + patch?

Thanks in advance.

kometa-triatlon, 2018-01-07:
Also, could you clarify how to patch the Jetson exactly? I am confused about the following statement: 

> Copy .dtb files in the current directory to the following directories:

If under the current dir you mean the folder where the `JetPack-L4T-3.1-linux-x64.run` resides, I cannot see any .dtb files. I can find some under 64_TX:

```
./64_TX2/Linux_for_Tegra_tx2/rootfs/boot/tegra186-quill-p3310-1000-c03-00-dsi-hdmi-dp.dtb
./64_TX2/Linux_for_Tegra_tx2/rootfs/boot/tegra186-quill-p3310-1000-c03-00-base.dtb
./64_TX2/Linux_for_Tegra_tx2/rootfs/boot/tegra186-quill-p3310-1000-a00-00-base.dtb
./64_TX2/Linux_for_Tegra_tx2/kernel/dtb/tegra186-quill-p3310-1000-c03-00-dsi-hdmi-dp.dtb
./64_TX2/Linux_for_Tegra_tx2/kernel/dtb/tegra186-quill-p3310-1000-c03-00-base.dtb
./64_TX2/Linux_for_Tegra_tx2/kernel/dtb/tegra186-quill-p3310-1000-a00-00-base.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/tegra186-a02-bpmp-quill-p3310-1000-c01-00-te770d-ucm2.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-c01-00-te770d-ucm2.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-c01-00.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-c04-00.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-c00-00.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-c00-00-te770d-ucm2.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-a00-00-te770d-ucm2.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-c04-00-te770d-ucm2.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/t186ref/tegra186-a02-bpmp-quill-p3310-1000-a00-00.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/tegra186-quill-p3310-1000-c03-00-base.dtb
./64_TX2/Linux_for_Tegra_tx2/bootloader/tegra186-bpmp-quill-p3310-1000-a00-00.dtb

```

So, I can see some .dtb files in the directories where they are supposed to be copied. Am I misunderstanding something?

Alexey-Kamenev, 2018-01-08:
To your last question: the current directory is [this directory](../tree/master/tools/install/j120-tx2-patch). 
However, we now recommend applying [latest patch](https://auvidea.com/firmware/) from Auvidea website instead. 

As for JetPack 3.2: we haven't tested it yet so I cannot say whether the issue has been fixed in that particular version of dev preview release.

kometa-triatlon, 2018-01-08:
@Alexey-Kamenev Thanks for the hint. 
Can you clarify also the following:

- Should I apply only the latest TX2 patch (1.5) or all the patches starting from 1.2?
- The readme file for 1.5 contains instructions for making USB working and updating the device tree. Should I execute both? Sequentially or these can be done simultaneously? 
- When patching, Jetson must be mounted on the original carrier board or on the J120?

Many thanks in advance

Alexey-Kamenev, 2018-01-08:
- the patches should be cumulative, so apply only the latest one (1.5)
- as I understand, you need to execute both steps according to the instructions. However, if that does not work, I suggest opening an issue on Auvidea [support page](https://auvidea.com/my-tickets/). 
Note about our patch: it fixes USB 2.0 but does not enable USB 3.0 but if you don't use USB 3.0 you can still try it as it seems to be a bit easier to apply.
- either way should be fine. The patch fixes a bunch of issues specific to J-120 and should not affect how Jetson works on dev kit board.

closed 2018-01-16 by kometa-triatlon

yangyangcv, 2018-04-13:
just verified, usb port on J120 with Jetpack3.2 doesn't work. don't know whether or not this patch will work