NVIDIA-AI-IOT/redtail #36: Linking stage error

mmcctt00, 2018-01-15:
The next error ocurred while running the jetson_ros_install.sh script on a TX2:
[ 73%] Built target camera_info_manager
[ 76%] Linking CXX executable /home/nvidia/ws/devel/lib/camera_info_manager/unit_test
CMakeFiles/convert.dir/src/convert.cpp.o: In function `main':
convert.cpp:(.text+0x134): undefined reference to ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
convert.cpp:(.text+0x2cc): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
convert.cpp:(.text+0x408): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
/home/nvidia/ws/devel/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::Emitter::Write(std::string const&)'
/home/nvidia/ws/devel/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::ostream_wrapper::write(std::string const&)'
/home/nvidia/ws/devel/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::node_data::empty_scalar'
/home/nvidia/ws/devel/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::node_data::set_scalar(std::string const&)'
/home/nvidia/ws/devel/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::Emitter::PrepareIntegralStream(std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >&) const'
collect2: error: ld returned 1 exit status

Any idea of what the problem could be 
Thanks

Alexey-Kamenev, 2018-01-16:
I cannot repro it so here is a couple of things you can try:
1. Make sure you are using JetPack 3.1 JetPack 3.2 is currently not supported.
2. Try installing `image_common` components via APT instead of building from source. See line 93 in  `jetson_ros_install.sh` script. You would need to comment out/remove lines 94-105.
3. Do you have any other components that are installed outside of redtail?

mmcctt00, 2018-01-16:
Just pointgrey's flycapture. I'll try flashing my Jetson and running your script before instaling anything else. Thanks for answering!

closed 2018-01-16 by mmcctt00