NVIDIA-AI-IOT/redtail #38: Auvidea J120

arybalkin, 2018-01-19:
have a quick question about the carrier board you used in  this project... I also have Auvidea J120 and cannot get CAN Bus working with TX2. Looking at modifications to device tree and to the mcp251x driver Auvidea made, it looks like it's all TX1 specific and I don't see they ever publish any changes specific to TX2 module. Can you share the information what you have to do to get CAN Bus working in your project with Auvidea J120 carrier and TX module?

Alexey-Kamenev, 2018-01-19:
In redtail we use UART, not CAN bus, so I'm afraid there is not much we can help with. I recommend opening an issue via [Auvidea support](https://auvidea.com/my-tickets/) page.

arybalkin, 2018-01-19:
Thanks, I was not sure. Did you have to keep Jetson't heatsink  on and have its fan running during inference?  I would love to save some weight, but in our environment TX2 runs quite warm in J120 even without nVME M2 card and at idle load.  The same TX2 module runs much cooler in the DevKit board. Unfortunately, Auvidea support is pretty much useless, IMHO.

Alexey-Kamenev, 2018-01-22:
Re: heatsink - as we were flying our drones outside we had a natural cooling, so we could remove the heatsink without the risk of overheating the Jetson. For rovers or indoor usage we would definitely recommend keeping the heatsink with fan. 

As for Auvidea: let me see what I can do here...

auvidea, 2018-02-01:
Sorry the MCP2515 driver only supports the TX1. We still struggling to port it to the TX2. So for the TX2 please use the native CAN port. The J120 with 2 CAN connectors supports 2 CAN interfaces.
J9 (CAN1) is the native port (TX2 only). J10 (CAN2) is the MCP2515 CAN port (TX1 only this time). With some patches J10 could be routed to the second native CAN controller of the TX2, if you need 2 CAN interfaces with the TX2. Please note that only the recent revisions of the J120 have both CAN connectors populated. 

Please check out the TX2 firmware for the J120 at auvidea.com/firmware.

arybalkin, 2018-02-01:
Thank you for clarifying the CAN situation. I have a J120 Rev.7 board and I can see that both CAN ports are populated with TJA1051TK/3 transceivers on that board. Can you let me know which TX2 interrupt I can use for the MCP2515 controller? If CAN2 can be routed to TX2 internal CAN controller on this board, can you share the device tree modification for this change?

arybalkin, 2018-02-18:
Any changes/updates on CAN support for TX2 on Auvidea J120 rev.7 boards? Are you planning to support MCP2515 with TX2 module? If not, can you provide patches to use J10 (CAN2) connector with native TX2 CAN controller (mttcan module)?

16houstbraj, 2018-08-28:
Has the software fix for this issue been published? Would like to know if I can get my other CAN port to work through J120 board with TX2.

viteo, 2021-02-26:
There is no software fix for that, because CAN2 is physically not connected to CAN GPIO on Jetson. You should desolder jumper from SPI line and add two jumpers (0 Ohm 0402) for CAN H and L lines. 
[image]
With this modification I'm able to use both CANs on rev.7 board.