NVIDIA-AI-IOT/redtail #44: Rover overcorrects steeering.  How to tune steering parameters?

TritonSailor, 2018-02-23:
This video shows driving behavior observed during initial testing of redtail running on a rock crawler platform.  The steering appears to be overcorrecting.

https://www.youtube.com/watch?v=Lexi6v5XSSQ

Before testing I attempted to adjust the linear_speed_scale and turn_angle_scale to normalize controller output to expected servo output values.  Is this the correct approach?  Could you please provide more info on setting up controller parameters?

These are the parameters that were used for running the controller node.

rosrun px4_controller px4_controller_node _altitude_gain:=0 _linear_speed:=2  _joy_type:="xbox_wired" _obj_det_limit:=0.3 _vehicle_type:=apmrover _linear_speed_scale:=200 _turn_angle_scale:=-400 _dnn_turn_angle:=45.0 _dnn_lateralcorr_angle:=45.0

Alexey-Kamenev, 2018-02-27:
Great video - thanks for sharing!

Yes, what you are doing is correct - at the moment, manually tuning these parameters (*_scale and *_angle) is the only way to tune for a particular rover/DNN/environment.
The reason for such rudimentary control is lack of good visual odometry implementation for the rover, and, as a result, inability to use simple and efficient waypoint episodic control that we use in the drone scenario. One quick idea that we had in the past was putting PX4Flow module on a stick attached to the rover but we never had time to actually test this idea. Since your rover is driving outside, you may try using GPS signal for localization and waypoint control but you may run into issues with accuracy and reception of GPS signal.

All that being said, we hope (and work) to improve things in the future so stay tuned.

nsmoly, 2018-03-06:
We hope to add a visual odometry/SLAM in the future release, so you will be able to use trajectory/waypoint control, which will avoid oscillations.