NVIDIA-AI-IOT/redtail #45: Missing px4_controller from rospack

isakdiaz, 2018-02-27:
I am trying to set up my xbox wireless controller to interface with gazebo in docker but I get the error: 
"[rospack] Error: package 'px4_controller' not found".

Everything else in the installation seems to run fine but "rospack list" command does not return the px4_controller as part of ros. Using "rostopic echo /joy" does show the controller is working in the docker just fine. According to the installation instructions, the quad should take off setting up the joystick but it doesn't do that either. Any suggestions?

Alexey-Kamenev, 2018-02-28:
Did you use our [redtail Docker image/container](../tree/master/tools/simulation)? If so, did you follow the instructions from [our wiki](../wiki/Testing-in-Simulator#redtail-docker)? Make sure you do `catkin_make` from `~/ws` and check `~/ws/src` directory to see if all components are there. Here is what I get in my docker:
```
root@n:~# rospack list | grep -i px4_controller
px4_controller /root/ws/src/px4_controller
root@n:~# ll ~/ws/src/
total 8.0K
drwxr-xr-x 2 root root 4.0K Jan 25 22:03 ./
drwxr-xr-x 8 root root 4.0K Jan 25 22:22 ../
lrwxrwxrwx 1 root root   50 Jan 25 21:28 CMakeLists.txt -> /opt/ros/kinetic/share/catkin/cmake/toplevel.cmake
lrwxrwxrwx 1 root root   19 Jan 25 21:28 angles -> /root/angles/angles/
lrwxrwxrwx 1 root root   40 Jan 25 22:03 caffe_ros -> /data/src/redtail/ros/packages/caffe_ros/
lrwxrwxrwx 1 root root   11 Jan 25 21:28 gscam -> /root/gscam/
lrwxrwxrwx 1 root root   45 Jan 25 22:03 px4_controller -> /data/src/redtail/ros/packages/px4_controller/
```

isakdiaz, 2018-02-28:
Yes I was using the redtail docker image and going straight off the wiki. The docker container however didn't have the redtail repo on the image itself so it couldn't find the packages.  After downloading the repo, catkin was able to make the packages correctly.

closed 2018-02-28 by isakdiaz