NVIDIA-AI-IOT/redtail #46: docker image not seeming to allow gazebo to run with nvidia-docker2

damien022, 2018-03-05:
I suspect it is a similar issue as the lack of opengl support with nvidia-docker2. [https://github.com/NVIDIA/nvidia-docker/issues/534](url)

vijay609, 2018-03-05:
Can you check if your driver version matches on the host and inside the
container? If you not you may have to install the correct driver version
from inside the docker container.

Vijay

damien022, 2018-03-06:
nvidia 390.30 installed on both host and container. When trying to launch gazebo I get libGL errors
> libGL error: No matching fbConfigs or visuals found
> libGL error: failed to load driver: swrast
I can run gazebo and other opengl gui applications when basing a container of nvidia/opengl:runtime

Alexey-Kamenev, 2018-03-06:
We are using `nvidia-docker` (v1) which works fine. My guess would be `nvidia-docker2` is doing something different which causes this problem to appear. If you can, try running gazebo using `nvidia-docker`. Be aware that containers created by v1 and v2 are not compatible.

damien022, 2018-03-06:
yes, nvidia-docker2 does not load the opengl drivers by default. starting the image from nvidia/opengl:runtime allows the running of gazebo / rviz with a vanilla kinetic / 16.04 installation. I will try and see if i can build a redtail image over the coming few days with nvidia-docker2. 
I think nvidia-docker1 is being deprecated, so would be worth making it all work with v2...

kidtronnix, 2018-09-25:
The correct line is... `FROM nvidia/cudagl:9.0-devel`

daylanc, 2018-10-03:
@kidtronnix What do you mean by "the Dockerfile.kinetic should simply build from `cudagl:devel`"? Do I have to run the container differently from the wiki: `run_redtail_docker.sh [container_name] [host_data_dir] [container_data_dir]` ?
Also, what do you mean by "The correct line is... `From nvidia/cudagl:9.0-devel`"?

daylanc, 2018-10-03:
@Alexey-Kamenev For the simulation, do I need to install Cuda 9.0?
Currently I have the following installed:
- NVIDIA Driver (version: 396.54)
- NVIDIA Docker (version: 2.0.3)
I also have the `TensorRT 4.0.1.6 for Ubuntu 1604 and CUDA 9.0 tar package` downloaded onto my computer. Do I need to install/download anything else to run the simulations properly?

damien022, 2018-10-04:
@daylanc You need to replace the first line in redtail/tools/simulation/docker/Dockerfile.kinetic from
`FROM nvidia/cuda:9.0-devel-ubuntu16.04`
with 
`FROM nvidia/cudagl:9.0-devel`
Then run the script tools/simulation/docker/build_redtail_image.sh as per the docs
cudagl adds the opengl libraries to the image, which means you can run gazebo in the container

Alexey-Kamenev, 2018-10-04:
CUDA is required in case you want to run full simulation which includes running TrailNet via `caffe_ros` package. TrailNet uses TensorRT, cuDNN and CUDA.
You can, however, run just the controller node, `px4_controller`, along with Gazebo without any CUDA dependency as mentioned in the [wiki](../wiki/Testing-in-Simulator#nvidia-gpu-and-cuda-support-optional) (that section of the wiki describes manual build of the container which is recommended only if default redtail Docker image does not suit your needs)

daylanc, 2018-10-05:
[image]
When attempting to build the docker image via `sudo ./build_redtail_image.sh /path/to/tensorrt/tar/file>` I run into the error above. I updated pip to 18.0, but it's still stating that I'm using pip 8.1.1
Here are some sanity checks I ran:
`$ sudo python --version`
`Python 2.7.12`

`$ sudo pip --version`
`pip 18.0 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)`

I don't know why it's still saying I'm using pip 8.1.1..

commit ac98489d (2018-10-11): Fixed #46.
closed 2018-10-11 by Alexey-Kamenev