NVIDIA-AI-IOT/redtail #47: Validation accuracy remains below 60%

TritonSailor, 2018-03-19:
Hello, I'm trying to verify I get correct results training with the forest trails dataset, using the instructions in the wiki, before I try to fine tune with my own data. However, I haven't been able to get validation accuracy above 60%.  

I tried running 40 epochs, without much difference.  Here's a few screenshots and a caffe log.  What other info can I provide to help diagnose this issue?
  

[image]
[image]
[caffe_output(5).log](https://github.com/NVIDIA-Jetson/redtail/files/1826060/caffe_output.5.log)

Alexey-Kamenev, 2018-03-20:
Hello,
60% validation accuracy is certainly low if you are training orientation head on original Forest Trails. With ResNet-18 you should easily get up to 85%. How did you create the dataset? I would guess you used our `idsia_trails_dataset_digits.py` with undersampling? The model that you are using looks good to me, it's the same that we used to train orientation head.
There are few options you can try:
* try creating larger dataset (oversampling)
* try using ordinary CE (cross entropy) loss instead of our custom loss by setting all parameters like `entScale` and others to 0 in `cee_loss` layer.
* try using a very small subset of your dataset to see if you can get higher accuracy.
* try changing other parameters, most notably minibatch size and learning rate.

If everything is setup correctly, here is what you should get (or something like that):

[image]

If nothing of that helps, I will try to re-train our nets again but this might not happen until April as we have GTC 2018 next week and currently a bit busy.

TritonSailor, 2018-03-23:
Thanks for your detailed response.  Turns out the underlying issue was a corrupted Forest Trails dataset zip file.  

I downloaded the sets individually, recreated the data set with oversampling, and ran the training again.  

Results now look as expected...

[image]

closed 2018-03-23 by TritonSailor

gochit, 2019-01-04:
Hi @TritonSailor , I ran into the same problem. My accuracy is only 50%. I tried to rebuild the dataset using  `idsia_trails_dataset_digits.py`, but nothing change. How did you solve the problem?

BTW, I have 1 Titan X and it took me 10 hours to complete 20 epochs. Is it normal?

Thank you.