NVIDIA-AI-IOT/redtail #50: Calibrating IRIS sensros

mkaidrone, 2018-04-16:
After updating firmware to the version listed in Wiki, it said  "Follow the standard steps to calibrate the sensors, RC, and flight modes for the drone.". I have tried to do this with both QGroundControl and Mission Planner. They both get stuck after calibrating accelerometer. Is there any workaround for this? Which one should be used for calibration? QGroundControl or Mission Planner?
Thanks!

Alexey-Kamenev, 2018-04-16:
Just to make sure, the firmware version that you are using is 1.4.4? What is the version of QGC that you are using and which platform (e.g. Ubuntu 16.04 etc) are you running it on? When you say "stuck", does it mean the calibration never finishes?

mkaidrone, 2018-04-18:
I am able to calibrate after downgrading the QGC version to 3.2. So i am closing this issue.

closed 2018-04-18 by mkaidrone