NVIDIA-AI-IOT/redtail #51: caffe digits ERROR: error code -11

zhong-xin, 2018-04-22:
When training orientation head,I met the following questions.

Setting up train-data
Top shape: 64 3 180 320 (11059200)
Top shape: 64 (64)
Memory required for data: 44237056
Creating layer data_aug

##System configuration
    Operating system: ubuntu 16.04
    Compiler:
    CUDA version (if applicable): 9.0
    CUDNN version (if applicable): 9.0.176
    Python version (if using pycaffe): tried both 2.7 and 3.5.2

Alexey-Kamenev, 2018-04-24:
What is the GPU that are you using? Try reducing minibatch size (e.g. to 8) and see if it works.

zhong-xin, 2018-04-25:
@Alexey-Kamenev Titan XP.Reducing minibatch size is not help.

zhong-xin, 2018-04-25:
@Alexey-Kamenev 
I0424 22:31:13.564702  1215 upgrade_proto.cpp:1084] Attempting to upgrade input file specified using deprecated 'solver_type' field (enum)': /home/xin/digits/digits/jobs/20180424-223112-f147/solver.prototxt
I0424 22:31:13.565052  1215 upgrade_proto.cpp:1091] Successfully upgraded file specified using deprecated 'solver_type' field (enum) to 'type' field (string).
W0424 22:31:13.565063  1215 upgrade_proto.cpp:1093] Note that future Caffe releases will only support 'type' field (string) for a solver's type.
I0424 22:31:13.565243  1215 caffe.cpp:204] Using GPUs 0
I0424 22:31:13.586935  1215 caffe.cpp:209] GPU 0: TITAN Xp
I0424 22:31:14.909298  1215 solver.cpp:45] Initializing solver from parameters:
test_iter: 164
test_interval: 3482
base_lr: 0.001
display: 79
max_iter: 34820
lr_policy: "poly"
power: 1
momentum: 0.9
weight_decay: 1.0000001e-05
snapshot: 8705
snapshot_prefix: "snapshot"
solver_mode: GPU
device_id: 0
net: "train_val.prototxt"
train_state {
level: 0
stage: ""
}
type: "Nesterov"
I0424 22:31:14.909458  1215 solver.cpp:102] Creating training net from net file: train_val.prototxt
I0424 22:31:14.910451  1215 net.cpp:294] The NetState phase (0) differed from the phase (1) specified by a rule in layer val-data
I0424 22:31:14.910498  1215 net.cpp:294] The NetState phase (0) differed from the phase (1) specified by a rule in layer accuracy
I0424 22:31:14.910944  1215 net.cpp:51] Initializing net from parameters:
state {
phase: TRAIN
level: 0
stage: ""
}
layer {
name: "train-data"
type: "Data"
top: "data"
top: "label"
include {
phase: TRAIN
}
data_param {
source: "/home/xin/digits/digits/jobs/20180424-221713-4a74/train_db"
batch_size: 64
backend: LMDB
}
}
layer {
name: "data_aug"
type: "Python"
bottom: "data"
bottom: "label"
top: "data"
top: "label"
include {
phase: TRAIN
}
python_param {
module: "digits_python_layers"
layer: "TrailAugLayer"
param_str: "{\'debug\': False, \'hflip3\': True, \'blurProb\': 0.1, \'contrastRadius\': 0.2, \'brightnessRadius\': 0.2, \'saturationRadius\': 0.3, \'sharpnessRadius\': 0.3, \'scaleMin\': 0.9, \'scaleMax\': 1.2, \'rotateAngle\': 15, \'numThreads\': 32}"
}
}
layer {
name: "sub_mean"
type: "Scale"
bottom: "data"
top: "sub_mean"
[... 1542 lines cut ...]
@     0x7f62486a9273 PyObject_Call
@     0x7f62486c9b75 (unknown)
@     0x7f6248660173 (unknown)
@     0x7f62486a9273 PyObject_Call
@     0x7f624864735c PyEval_EvalFrameEx
@     0x7f624878005c PyEval_EvalCodeEx
@     0x7f6248641da9 PyEval_EvalCode
@     0x7f62486e3244 PyImport_ExecCodeModuleEx
@     0x7f62486e3c1f (unknown)
@     0x7f62486e5390 (unknown)
@     0x7f62486e5658 (unknown)
@     0x7f62486e676b PyImport_ImportModuleLevel
@     0x7f62486508b8 (unknown)
@     0x7f62486a9273 PyObject_Call
@     0x7f624877f487 PyEval_CallObjectWithKeywords
@     0x7f62486457e6 PyEval_EvalFrameEx
@     0x7f624878005c PyEval_EvalCodeEx
@     0x7f6248641da9 PyEval_EvalCode
@     0x7f62486e3244 PyImport_ExecCodeModuleEx
@     0x7f62486e3c1f (unknown)
@     0x7f62486e5390 (unknown)
@     0x7f62486e5658 (unknown)
@     0x7f62486e676b PyImport_ImportModuleLevel
@     0x7f62486508b8 (unknown)
@     0x7f62486a9273 PyObject_Call

Alexey-Kamenev, 2018-04-25:
From the log, it looks like something went wrong when `data_aug` layer was being created. This layer is implemented using Python custom layer which is defined [here](../blob/master/models/nets/python-layers.py). Can you please check that you provide correct path to this file when you run a job in DIGITS and that this file is accessible by DIGITS (e.g. if you run DIGITS in Docker container). See [wiki](https://github.com/NVIDIA-Jetson/redtail/wiki/Training-TrailNet-model#training-orientation-head) for more details.

Alexey-Kamenev, 2018-05-07:
Hope you were able to fix the problem, but if not - feel free to re-open the issue or create a new one.

closed 2018-05-07 by Alexey-Kamenev