NVIDIA-AI-IOT/redtail #54: UART of Auvidea J120 with TX2

yangyangcv, 2018-05-02:
maybe it's not suitable to raise an issue here. but really i cannot find another place to discuss J120+TX2. I want to use the UART of Auvidea J120 and TX2. but i don't know physically where is the uart port on J120. besides, i see /dev/ttyTHS1, /dev/ttyTHS2, /dev/ttyTHS3. don't know which port could I use? thanks so much if anyone could give a hint.

Alexey-Kamenev, 2018-05-02:
You are welcome to ask questions here, we'll try to help.
As for UART on J120 you may check out the following documents:
* [J120 Manual](https://auvidea.com/download/manual/J120/J120_technical_reference_1.9.pdf) from Auvidea. In particular, look at page 6 - you will see the location of UART 0/2 (which is J7 connector). One thing to keep in mind is our build was tested with J120 Rev 4. In Rev 6 there were some changes to this port - make sure to read page 19 for details. If possible, I would recommend using Rev 2-5 or 7. If you still have questions or concerns then I recommend contacting Auvidea support.
* Relevant parts of our assembly guides:
  * [TBS Discovery](../wiki/Skypad-TBS-Discovery-Setup#3-control-from-j120-to-pixfalcon)
  * [IRIS+](../wiki/3DR-Iris-Setup#jetson-to-pixhawk-wiring)

If everything is wired properly, UART port will be mapped to `/dev/ttyTHS1` by default.

yangyangcv, 2018-05-07:
thanks so much! sadly still not work. i guess there's something with my carrier board. thank you alexey

closed 2018-05-07 by yangyangcv

Alexey-Kamenev, 2018-05-07:
If you suspect something is wrong with J120 then you can try using Jetson DevKit board UART port (J17) - we documented it [here](../wiki/Dev-board-UART-and-Pixhawk-connection). This will allow you to test Pixhawk - Jetson connectivity. If it works with dev board and does not work with J120 then you probably should contact Auvidea support.