NVIDIA-AI-IOT/redtail #55: Dataset Preperation

earcz, 2018-05-02:
Hi,

I am interested in your work and want to thank for your sharing. Also, I am new to deep learning.
I have started to implement your project myself. Here are my questions:

1. I would like to create my own dataset. If i convert rgb images to grayscale, doest that affect training in a bad manner? because my gpu is GTX 1070. What do you think ?
2. As is well known, I need to resize images before training. What size should images be? I mean how can I decide about size of images? According to the network what I will use or something else?

Kind regards, Ender.

earcz, 2018-05-02:
Thank you for the response. I got it now. However, because of your response, I have an another question which is that do I have to use DIGITS Docker? Because I am not using docker.

isakdiaz, 2018-05-02:
Well docker is the recommended way of using DIGITs which simplifies the process. It's not necessary. You just need to come up with the prototxt and caffemodel files and put them in the "/home/nvidia/redtail/models/pretrained" folder. Once it's there, you just put them as the parameter to the caffe_ros node when you run it.

closed 2018-05-03 by earcz