NVIDIA-AI-IOT/redtail #56: Where to Start

earcz, 2018-05-03:
I think this is the most general question in here, however I need to ask with all due respect. It is an amazing open source project. I dedicated myself to fork this project with your guidance. For a while, I am inspecting this project and surveying literature. I also bought jetson tx2, new computer with GTX 1070 and got system ready to start. Unfortunately, I am all alone for this project and struggling where to start. I just need right start point and a spark. Would you please give me that spark?

Thank you in advance, Ender.

Alexey-Kamenev, 2018-05-04:
I would recommend starting with setting up and running our code in [simulator](../wiki/Testing-in-Simulator). This will help you to understand how the whole system works, how components interact with each other, and, last but not least, to have fun by flying the drone (not a real drone, of course, but still pretty cool). That way you won't have to spend time/money on building hardware (which is fun, but not exactly an easy thing to do) or setup a training pipeline (again, fun thing to do, but probably should not be the first item on your list - use our [pre-trained models](../tree/master/models/pretrained) instead).

Note that we use Docker extensively and strongly recommend doing as much work as possible from Docker containers (that includes everything from running Gazebo simulator to code development using [Visual Studio Code](https://code.visualstudio.com/)). If you are not familiar with Docker, I encourage you to spend some time and become familiar with it - it will save you a lot of time in the future!

Good luck and feel free to ask questions here.

earcz, 2018-05-04:
Thank you for your response. Running your code in simulator is good for a fresh start and I started in that way. I want to say that I have already built a quadcopter for a competition so that after redtail simulation experience, I will have chance to implement it in real life. That is why I have bought Jetson TX2 and a computer to train my own dataset. As you said, I have encountered Docker in simulator page of redtail when I have started yesterday. That is really strange to me as a mechanical engineer but I understood that I need to get familiar and started to learn it.

**Sincere gratitude for your continuous support.**

closed 2018-05-04 by earcz