NVIDIA-AI-IOT/redtail #59: PX4 and ROS Docker Setup Issue

earcz, 2018-05-06:
**Hi,**

Because, unfortunately, I could not use Redtail docker image although I have tried a few times, I have started to build image manually regarding the guide, starting from **PX4 and Ros Docker setup (step 2)**. I have followed the steps on [PX4 developer page](https://dev.px4.io/en/test_and_ci/docker.html#manual_start) by starting with this:
```
mkdir src
cd src
git clone https://github.com/PX4/Firmware.git
cd Firmware
```
To pull px4-dev-ros:v1 , I implemented "Calling Docker Manually" section as follows:
```
# enable access to xhost from the container
xhost +
```
-  First problem showed up with `xhost +` command and output: _access control disabled, clients can connect from any host_. I googled that but could not find solution although I have found a recommendation but did not help. Nevertheless, I kept going on running docker as follows:
```
#Run docker
docker run -it --privileged \
    --env=LOCAL_USER_ID="$(id -u)" \
    -v ~/src/Firmware:firmware:rw \
    -v /tmp/.X11-unix:/tmp/.X11-unix:ro \
    -e DISPLAY=:0 \
    -p 14556:14556/udp \
    --name=px4-ros px4io/px4-dev-ros:v1.0 bash
```
- However, output turned out: **docker: Error response from daemon: invalid volume specification: '/home/deep/src/Firmware:firmware:rw': invalid mount config for type "bind": invalid mount path: 'firmware' mount path must be absolute.**
I also tried Helper Script (docker_run.sh) but nothing was okay. Could you please help me?

Kind regards, Ender.

Alexey-Kamenev, 2018-05-07:
Is it possible to share more details on what did not work when you tried building redtail Docker image?

The warning that you got after running `xhost` does not mean something is not working - it is a warning that any client now can connect to your X server (`+` command line parameter). This is needed when the container is first created by `docker run` command so you can run GUI apps from the container itself. Once the container is created, each subsequent runs  (via `docker start` command) use the container's host name to grant permissions to X server only for that container. See more details on `xhost` [here](https://www.x.org/archive/X11R6.8.1/doc/xhost.1.html).

Your second problem is just an incorrect argument to `-v` option of `docker run` command: as it was stated in the error message, all paths should be absolute. So in your case it should be something like `-v ~/src/Firmware:/src/firmware:rw`. In this case, your `~/src/Firmware` host directory will be mapped into `/src/firmware` directory in the container. [More info](https://docs.docker.com/engine/reference/commandline/run/#mount-volume--v---read-only).

In general, the main reason to use Docker is to isolate your host machine from changes in configuration made by different packages that have [complex dependencies](https://en.wikipedia.org/wiki/Dependency_hell). You may choose not to use Docker and it may even work for some time but eventually your chances of running into some tricky dependency issue will be quite high. In redtail project, the only place where we don't use Docker is on Jetson itself - re-flashing it is simple and our [Jetson install script](../tree/master/ros/scripts) takes care of all required dependencies.

earcz, 2018-05-07:
When I run `./build_redtail_image.sh /Home/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz`,  it gives output as:
```
Using -v2 image suffix.
cp: cannot stat '/Home/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz': No such file or directory
unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/deep/Downloads/Dockerfile.kinetic: no such file or directory
```
I have googled to understand why this is happening but could not find anything.

Regarding my second problem, you are right. It is my fault. I have changed it as you stated and It just run now but with the output of: 
```
Cloning into 'jMAVlib'...
Submodule path '../../Tools/jMAVSim/jMAVlib': checked out 'afa488d166813038c28572907eacf6eaa54f3c7c'
-- Configuring incomplete, errors occurred!
See also "/src/firmware/build/posix_sitl_default/CMakeFiles/CMakeOutput.log".
/bin/sh: 1: cd: can't cd to /src/firmware/build/posix_sitl_default
make: *** [posix_sitl_default] Error 2
```
Afterwards, I have just run `docker start px4-ros`, It worked inspite of the error.

Alexey-Kamenev, 2018-05-07:
Does `/Home/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz` file exist? What do you see if you run:
```
ls /Home/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz
```

earcz, 2018-05-07:
Yes, it exists. Please see the screen shot below.  Still, `ls /Home/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz` gives output of: 

`ls: cannot access '/Home/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz': No such file or directory`

[image]

Alexey-Kamenev, 2018-05-07:
The path you see in File Explorer is not the actual path on your disk. Use the following to determine the full path:
```
ls ~/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz
```
In case you need a refresher on Linux command line, check out [this](http://linuxcommand.org/tlcl.php).

As for CUDA: if you are using Docker, you don't need it on the host (it's already in the container), however, if you decide to install everything on the host directly then you should also install all dependencies yourself.
You will need NVIDIA drivers in either case (CUDA comes with drivers but I recommend installing latest ones via apt/directly).

earcz, 2018-05-07:
`ls ~/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz` gives output of: 
`/home/deep/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz`

and then, I have downloaded tensorrt again to be sure that it is fully downloaded, and run the command again
`cd ~/Downloads`
`./build_redtail_image.sh /home/deep/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0`

Still, the same problem exists:
```
Using -v2 image suffix.
cp: '/home/deep/Downloads/TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz' and './TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-8.0.cudnn7.0.tar.gz' are the same file
unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/deep/Downloads/Dockerfile.kinetic: no such file or directory
```
Thank you for your patience.

Alexey-Kamenev, 2018-05-07:
How did you get redtail sources? You should clone the full repo, then navigate to appropriate directory and run scripts from there. Do not download separate files, this won't work.

earcz, 2018-05-08:
I got it now. It was my fault because I am new to Linux system. I appreciate for your support.

Glidepathuav, 2018-05-08:
It's not a silly question if you do not know the answer!

toreal, 2018-05-14:
I also got the same situation. How to solve the` <none> REPOSITORY and <none> TAG `problem?

I can not install nvidia-docker v1,  thus I install nvidia-docker2 instead. Did that make this problem?

`REPOSITORY          TAG                            IMAGE ID            CREATED             SIZE`
`<none>              <none>                         f65789ee0122        9 minutes ago       3GB `

earcz, 2018-05-14:
I don't think that installing  v2 is causing the problem because I have installed v1. So, the problem must be something else. I could not find the solution for that. Today I will try to install images manually.

toreal, 2018-05-14:
I found [pip v10 issue ](https://github.com/pypa/pip/issues/5221).

Thus, I mark the line in Dockerfile.kinetic as follows.
`## && pip install --upgrade pip \`

This seems to work.

earcz, 2018-05-14:
Thank you, that worked for me, too !