NVIDIA-AI-IOT/redtail #61: Rover mode for Pixhawk Cube?

MrDadaGuy, 2018-05-09:
I'm trying to get a Rover working with Pixhawk cube and TX2.  

When I use your custom firmware, there's no option for Rover, only air frames. 

When I use APM firmware, there is an option for Rover, but there's no SYS_COMPANION parameter, and I'm not getting any status updates from MAVROS. 

Any ideas?

Thanks!!

rbachtell, 2018-05-09:
I think at this point we are out of luck as Redtail has not yet been ported to the APM stack and the object avoidance system from the APM developers has not yet been ported to the TX2.  It sounds like both are in the works but not yet released.

Cheers, RB

MrDadaGuy, 2018-05-09:
So, I've been playing around a lot, and now I can get messages back from pixhawk.  

I flashed the pixhawk cube with APM Rover firmware image.  I can get things going with this command:

`roslaunch mavros apm.launch fcu_url:="/dev/ttyACM0:921600"`

Then issuing the following command I can see compass heading get updated.

`rostopic echo /mavros/global_position/compass_hdg`

Alexey-Kamenev, 2018-05-10:
There is currently a very limited support for APM rovers in redtail. Few things to mention:
* make sure you are using latest stable version of [APM Rover firmware](http://ardupilot.org/rover/docs/common-downloads_firmware.html)
* if applicable, use proper [parameters](https://github.com/ArduPilot/ardupilot/tree/master/Tools/Frame_params) for your frame. 
* configure [TELEM2 port](http://ardupilot.org/copter/docs/parameters.html#serial2-protocol-telemetry-2-protocol-selection) on your Cube to work with TX2.

`px4_controller` currently supports so-called RC override mode for rovers which is essentially a direct control of the motors which is a hack. The proper way is to use waypoints, same as in case of the drones. The reason we could not do it on the rover is various APM limitations which hopefully will be fixed in future releases. Internally we use [AION R1 rover](https://www.aionrobotics.com/r1) with our own custom version of APM as well as VO system. 

For those who are interested in RC override version, we published a [.launch file](../blob/master/ros/packages/px4_controller/launch/aion_ctl.launch) for R1 which demonstrates how to run `px4_controller` in RC override mode in APM rover.

Alexey-Kamenev, 2018-09-26:
Closing, feel free to re-open or create a new one.

closed 2018-09-26 by Alexey-Kamenev