NVIDIA-AI-IOT/redtail #65: Redtail Image Building Errors

earcz, 2018-05-16:
Hello,

- While building redtail image, there exists lots of errors, such as `Failed building wheel for px4tools`, `Traceback` error is related to `no module named 'numpy'` , some performings tests Failed, and so on _(you may want to see images below)_

- After creating redtail container with `run_redtail_docker.sh` and before building components with `~/build_redtail.sh /data/src/redtail` , do we need to install something else? (to unzip files and etc.) 

- caffe ros node cannot be installed

- Although the errors and failed tests, px4 node and gazebo runs. However, in the terminal for entering `commander takeoff` command, there produces lots of warning messages. Is that normal?

**Screen shots to see the errors while building redtail image:**
[image]
[image]
[image]
[image]
Best regards, Ender.

Alexey-Kamenev, 2018-05-16:
The warnings that you see can be ignored.
As for `build_redtail.sh` script error, make sure you have provided full path to your redtail repo (e.g. in your case `/data/src/redtail` should exist and contain redtail GitHub code). As mentioned in the [wiki](../wiki/Testing-in-Simulator#building-components), our Docker script does not automatically clone redtail repo to avoid unnecessary duplication of the code.

BTW, I've just updated redtail Docker file, the image build process should now be working fine. You don't have to do anything as you already built an image which seems to work.

Alexey-Kamenev, 2018-05-17:
Note that the command examples are just **examples** - you should always verify that such commands make sense for your machine/configuration. For example, you may have redtail sources in a directory other than `/data/src/redtail`.
As for joystick type: AFAIR, the button/axis mapping on XBox One is the same as on Shield so you can use default joystick type (`shield`) or try other types (supported: `shield`, `xbox_wired`, `xbox_wireless`, `shield_2017`). The only difference in all those types is button mapping.
Also, make sure you are passing correct joystick device to `joy` node (e.g. `_dev:=/dev/input/js0`). Depending on you machine, you may have more than one joystick device, for instance, some mouse devices also act as joystick.

As for the links: most likely you did not provide correct path to `build_redtail.sh` script. You should delete those broken symlinks and re-run the script again (see the script itself for more details on what it does).

earcz, 2018-05-17:
MAVROS, Gazebo and QGroundControl are running well. I can use joystick either. Thank you!

Last but not least, I want to be sure If I did make a mistake at the beginning. I cloned redtail repo into Home directory. When [creating container](https://github.com/NVIDIA-Jetson/redtail/wiki/Testing-in-Simulator#creating-container) with respect to the explanation for default settings (`container_name`, `host_dir` and `container_data_dir`), I have just run `run_redtail_docker.sh` without defining any directory and container name. Afterwards, in the container, I have run `~/build_redtail.sh /data/src/redtail`. Is that wrong?

earcz, 2018-05-18:
As you said, it is because I of wrong path. I provided accessible path from the container. Everything is running now. 
BTW, there are another broken links. Should I symlink each other or delete them ? as is seen below
[image]

closed 2018-05-28 by earcz