NVIDIA-AI-IOT/redtail #67: Cannot control drone in gazebo

earcz, 2018-05-28:
Hi,

It is impossible to control the drone in Gazebo with X-Box One controller. For example, when i push a little into the button and keeping my finger on it for take off command, the drone takes off continuously. When i release it a little, it just drops. I could not make it to control. I have calibrated the controller from QGroundControl and set the buttons for related movements. However, nothing was changed. Is it about the controller or something else? 

Also, excluding calibration and button settings , I cannot switch settings on QGC. For example, I have a few times tried to change my battery cells, It did not apply. There are lots of problem with QGC. What can I do?

Best regards, Ender.

Alexey-Kamenev, 2018-06-01:
If you follow the steps from the wiki and everything is configured and running properly, then you should be able to control the drone with the joystick. Once the `px4_controller` is running, the drone should take off and after reaching target altitude (default is 2m) it should just hover. Now you can control the drone with the joystick. If something is not working - check the logs, outputs of all components and check the wiki one more time.

As for QGC: if you are trying to change parameters for the simulated drone (i.e. in Gazebo) then it may not work as that model does not support all parameters of the real drone. If you are having problems configuring a real drone with QGC then I would recommend checking out [QGC manual](https://docs.qgroundcontrol.com/en/).

earcz, 2018-06-01:
For this project, I have followed wiki strictly. So, as for `px4_controller`, When i run it, drone never takes off automatically. I am getting the drone armed and flying. Moreover, It cannot be controlled. Either it ascends or drops. Would you recommend me to set up the entire system from the very beginning ? (I ,in fact, did it a few times).

earcz, 2018-06-02:
Solved by rebuilding redtail image and container.

closed 2018-06-02 by earcz