NVIDIA-AI-IOT/redtail #68: Failed to parse camera calibration

earcz, 2018-05-31:
Hi,
I have calibrated my GoPro Hero4 camera in the host. There was created `calibration.yml` file. I changed its name as `camera.yml` and copied it into `/root/.ros/camera_info/`. By the way, at first, there was a file which is camera_info but it was not a directory. Thus, I deleted it and mkdir `camera_info` Then I have run the code of `rosrun image_pub image_pub_node _img_path:=/data/trail_test.mp4`. It gave the error in the image below. 
[image]

I am getting this error for two days although I have tried a few alternative solutions. I could not find a way out for this issue. I kindly request that would you please give me a hand?

Best regards, Ender.

isakdiaz, 2018-06-01:
The original error is a warning that there is no calibration file. You can get away with that and have it run NN just fine. If you try to create a calibration file then it will crash in error. It's possible to make a calibration file, but I had no luck getting the formatting correct. The easiest thing is just to delete it.

earcz, 2018-06-01:
I have deleted it and tried to run DNN. However, It still gave the error of camera.yml not found. After that, It gave the error for DNN like this: 
[image]

I am just curious that were you be able to run DNN without problem?

isakdiaz, 2018-06-01:
Are you sure both the model file paths are correct?

isakdiaz, 2018-06-01:
You will have to follow the instructions in the wiki.

Plug in camera, run gscam node, verify output of camera/camera_raw topic. Google how to show an image from gscam.

Once you know your image is working. Type in "rostopic list" to see all the topics running. Then use  "rostopic echo" command to print the output of your trails_dnn topic.

This should change as you move your camera around, but if you want to see the turning angle then you have to run the pc4controller node, and for this node you will have to have a the joystick in order to activate the DNN.

earcz, 2018-06-01:
Thank you, I will work on it.

Alexey-Kamenev, 2018-06-01:
As for calibration: see details [here](#52). Isak provided good list of steps to follow in the [earlier response](#issuecomment-393900814). You can also use `image_pub` node as described in the wiki.

earcz, 2018-06-01:
Link in "As for calibration: see details here" goes to this issue that I have opened. 
I have followed the steps that Isak provided and in wiki. I skipped the camera calibration step. With the image_pub, I defined `_img_path:=/dev/video1` to get logitech usb camera working (gopro hero4 does not work). I have checked `rostopic list ` and `camera/camera_raw` was running. Afterwards, I have run the DNN node, It has also run. I have checked it with `rostopic echo /trails_dnn/network/output` and it gave the output as you provided in wiki. However, although everything is running successfully, drone is not hovering in gazebo.

Alexey-Kamenev, 2018-06-01:
Updated the comment with correct link.

What is the console output of `px4_controller` node?

earcz, 2018-06-01:
I am running gazebo, mavros and joy node respectively as mentioned in wiki. When I run QGC, It starts but mostly it is giving a warning as `partner id mismatch`. It usually happens but not always.

Just now, This is the first time I cannot run `joy_node`. So,I checked `ls /dev/input/js*` and there was no `js*`. However, QGC sees my controller. It just happened now. 

As for running Gazebo, console output in sequence:
[image]
[image]

As for running MAVROS, console output in sequence,
[image]
[image]

earcz, 2018-06-02:
I have rebuilt the system for a fresh start. Each node is running perfect now (Mavros, Gazebo, DNN, controller).

Since `image_pub` cannot be used because of calibration issue, I have run `gscam` to use Logitech USB webcam. It also did not work because of the same reason. So this time, I have run `gscam` to run my Laptop's on-board camera and it worked. I have opened [redtail demo video](https://www.youtube.com/watch?v=USYlt9t0lZY&t=117s) from my mobile phone and put that in front of the Laptop's webcam. Drone did not fly. I will try it on a real trail, in addition to solving calibration issue. 

Thank you.

earcz, 2018-06-04:
System is running perfect now. It was about the calibration file that the system cannot recognize it since `widecam_mono_calibration` gives different output (image below) compared to calibration file at [this link](https://github.com/csc301/iarc/blob/master/image_common/camera_calibration_parsers/camera.yml). Although the calibration information in the [link](https://github.com/csc301/iarc/blob/master/image_common/camera_calibration_parsers/camera.yml) does not belong to my camera, I copied it and paste into my `camera.yaml` file with some modifications (`image_width`, `image_height` and `camera_name`). It worked successfully. 

However, I would like to know whether I did not understand calibration logic of your system or that is the right thing that I have applied.

[image]

Alexey-Kamenev, 2018-06-04:
`image_pub` node does not require calibration file (i.e. it's optional). 
The issue with calibration file in particular and usage of calibration information in general was described in my response to #52 - please read it. If you need more information about camera calibration then Google is your best source (e.g. [this](https://docs.opencv.org/3.3.0/dc/dbb/tutorial_py_calibration.html)).

closed 2018-06-04 by earcz