NVIDIA-AI-IOT/redtail #71: Could not get gstreamer sample.

mattgorski, 2018-06-14:
I am trying to test the camera node and I have tried to adjust the variables and the gscam examples will launch but everything.lauch and command line launch fails with: 

Could not get gstreamer sample.

I can launch manually using instructions here from examples changing ffmpegcolorspace to videoconvert

https://github.com/ros-drivers/gscam/issues/25

mattgorski, 2018-06-14:
Well i bypassed the issue and im not even sure it is the correct way but here is what i changed in everything.launch

changed v4l2src to nvcamerasrc

<env name="GSCAM_CONFIG" value="nvcamerasrc

Alexey-Kamenev, 2018-06-16:
What camera are you using? If it's the one that comes with Jetson devboard then we recommend using some other (USB) camera instead.
You can get this error even with USB cameras though - usually in case the camera does not support the resolution specified in GStreamer pipeline (width, height). You can use [v4l2-ctl](https://www.mankier.com/1/v4l2-ctl) utility to get information about the camera.

mattgorski, 2018-06-22:
This was solved by using a standard v4l2 UVC camera instead of onboard camera.  You can close this issue.  Thanks for the absolutely fantastic Wiki!

closed 2018-07-02 by Alexey-Kamenev

ioarun, 2022-11-30:
For those of you who still get this error, make these changes in your `v4l.launch` file.  Set `GST10` argument to `true` and replace  ` <param if="$(arg GST10)" name="gscam_config" value="v4l2src device=$(arg DEVICE) ! video/x-raw,format=RGBx,framerate=$(arg FPS) ! ffmpegcolorspace"/>`  with this: `<param if="$(arg GST10)" name="gscam_config" value="v4l2src device=$(arg DEVICE) ! video/x-raw,framerate=$(arg FPS) ! videoconvert "/>`

PS -  Thanks to Michael @meee1 & Sid @bugobliterator to help me with this.