NVIDIA-AI-IOT/redtail #72: About Pixhawk and Camera

earcz, 2018-06-27:
As for autopilot according to the wiki, pixhawk-1 is used for 3DR Iris and Pixfalcon for Skypad TBS. I am curious about if I use [Pixhawk-2 Cube](http://ardupilot.org/copter/docs/common-pixhawk2-overview.html) on my drone that I have built, Is it okay with your system that is running on Jetson TX2? Do you have any suggestion about that?

As for camera, I want to use stereo camera of [Intel RealSense](https://realsense.intel.com) instead of USB webcam. What can you say about this camera to use (for performance and compatibility)? Do you have any experience or hearings about that? 

Best.

Alexey-Kamenev, 2018-07-02:
We use Cube with our [Aion R1](https://www.aionrobotics.com/r1) rover and it works just fine. As for the camera: we haven't tried RealSense so I don't know if it's going to work. On our drones/rovers we use either webcams or [ZED](https://www.stereolabs.com/) camera.

daylanc, 2019-03-04:
@earcz 
How did the Intel RealSense camera work for you? I'm looking around for materials and stuck on choosing the best/correct camera.

earcz, 2019-03-05:
I did not use Intel RealSense, but logitech c930e webcam. Also for a while, I am using ZED camera for my another project. It is the best camera I have ever used. If your budget is okay, you can use ZED through redtail project.

daylanc, 2019-07-23:
@earcz 
Was the logitech c930e webcam work with the jetson tx2 running ubuntu? On the spec sheet it looks like it requires windows.