NVIDIA-AI-IOT/redtail #73: Simulator on x64 Host

daylanc, 2018-07-16:
Hello, I interested in trying out this project and was looking up where to begin. I found an old issue #56 and @Alexey-Kamenev stated to start by running the simulator.  In the Simulator wiki, it states, "The current configuration assumes all components are installed on a x86 host platform running Ubuntu 16.04." Does this mean the Docker will only work on 32-bit platforms? Can a 64-bit host not be used to run the simulation?

Also, in order to run the simulation, do we need the hardware components (drone, camera, etc.) or are we able to test everything in simulation? I am still in the "research" phase of this project and would like to get clarity on some issues before starting this investment. 

Thank you!

Alexey-Kamenev, 2018-07-16:
Hello,

We only tested our simulator on x86-64 machines. x86 is the generic [name of the ISAs](https://en.wikipedia.org/wiki/X86) which includes [x86-64](https://en.wikipedia.org/wiki/X86-64) among many other instruction sets.  We should have made it more clear that we recommend 64-bit platform as anything else is not tested. 
BTW, starting JetPack 3.2 you can run Docker directly on Jetson but we did not have time to test this scenario either.

As for your second question: you don't need a drone or any drone-related hardware components (like Pixhawk etc) to run the simulator. The camera is also optional as you can feed a test image or video as an input to DNN using `image_pub` ROS node to simulate the camera. You will need a joystick though, such as XBox or NVIDIA Shield.

daylanc, 2018-07-17:
Thank you for the quick response @Alexey-Kamenev. I haven't worked with Gazebo and Mavros before. In your simulator tutorial, you mentioned "Note that default joystick type is NVIDIA Shield, if you have a XBox then pass the appropriate joy_type parameter to the px4 controller node." I do have an available XBox controller. What is the "joy_type" parameter that I need to give the px4 controller node? In other words, how would I edit this to run it with an XBox controller?:

cd ${CATKIN_WS}/
rosrun joy joy_node _dev:=/dev/input/js0 _autorepeat_rate:=30 &
rosrun px4_controller px4_controller_node _altitude_gain:=2

Alexey-Kamenev, 2018-07-19:
Gazebo has an extensive [documentation and tutorials](http://gazebosim.org/tutorials) which I recommend checking if you haven't used it before. MAVROS has detailed [documentation](http://wiki.ros.org/mavros) as well.

As for the `joy_type`: supported joystick types are `shield`, `xbox_wireless`, `xbox_wired`, `shield_2017`. They only differ in axis/button mapping as can be seen [here](https://github.com/NVIDIA-Jetson/redtail/blob/master/ros/packages/px4_controller/src/px4_controller.cpp#L543). So to use wired XBox controller, run the controller node with the following parameters:
```
rosrun px4_controller px4_controller_node _altitude_gain:=2 _joy_type:=xbox_wired
```

Alexey-Kamenev, 2018-08-29:
Closing the issue, feel free to re-open or create a new one.

closed 2018-08-29 by Alexey-Kamenev

daylanc, 2018-09-17:
@Alexey-Kamenev You mentioned using `image_pub` to simulate the camera when running gazebo. How can I feed a test image to bypass the camera?