NVIDIA-AI-IOT/redtail #74: cannot find -lnvstereo_inference.

rahulsharma11, 2018-07-26:
Hi, While building the inference code, it throws error of "cannot find -lnvstereo_inference".
Also not able to see "nvstereo_inference" in lib folder.
Please suggest.
Thanks.

Alexey-Kamenev, 2018-07-27:
How did you build the library? We recommend following the [instructions](../tree/master/stereoDNN#building-inference-code). If you build debug version by following the instructions, the static library will be located at `./build/lib/libnvstereo_inference_debug.a`. The sample app binary will be located at `./bin/nvstereo_sample_app_debug`.

rahulsharma11, 2018-07-28:
Ok thanks for clarification.
One more thing can you please tell me as i have converted the tenserflow model to tensorRT. Now for "Building Inference code" section, i have to do this on jetson itself or my laptop? As i am not clear from the documents . What is the exat output file to be moved to jetson to run.

rahulsharma11, 2018-07-30:
Hi, I am building with debug mode directly on jetson tx2 itself. While make, it throws error-
/usr/bin/ld: cannot find -lnvstereo_inference
How can i correct this?

rahulsharma11, 2018-07-30:
Hi am using jetpack3.2.1.
And followed cmake -DCMAKE_BUILD_TYPE=Debug ..
what you suggested above.
Isuue is while make.

Alexey-Kamenev, 2018-07-30:
Unfortunately, this does not repro on my Jetson TX2 with JetPack 3.2 and fresh clone of redtail.
```
nvidia@tegra-ubuntu:~$ git clone https://github.com/NVIDIA-Jetson/redtail.git redtail_1
Cloning into 'redtail_1'...
remote: Counting objects: 558, done.
remote: Total 558 (delta 0), reused 0 (delta 0), pack-reused 557
Receiving objects: 100% (558/558), 549.61 MiB | 29.81 MiB/s, done.
Resolving deltas: 100% (180/180), done.
Checking connectivity... done.
Checking out files: 100% (213/213), done.
nvidia@tegra-ubuntu:~$ cd redtail_1/stereoDNN/
nvidia@tegra-ubuntu:~/redtail_1/stereoDNN$ mkdir build
nvidia@tegra-ubuntu:~/redtail_1/stereoDNN$ cd ./build/
nvidia@tegra-ubuntu:~/redtail_1/stereoDNN/build$ cmake -DCMAKE_BUILD_TYPE=Debug ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Using _debug suffix.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-9.0 (found suitable version "9.0", minimum required is "9.0") 
-- Found OpenCV: /usr (found suitable version "3.3.1", minimum required is "3.3.1") 
-- Found GTest: /usr/src/gtest/libgtest.a  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia/redtail_1/stereoDNN/build
nvidia@tegra-ubuntu:~/redtail_1/stereoDNN/build$ make
[  5%] Building NVCC (Device) object lib/CMakeFiles/nvstereo_inference_debug.dir/nvstereo_inference_debug_generated_kernels.cu.o
Scanning dependencies of target nvstereo_inference_debug
[ 10%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/conv3d_transpose_plugin.cpp.o
[ 15%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/conv_utils.cpp.o
[ 20%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/slice_plugin.cpp.o
[ 25%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/elu_plugin.cpp.o
[ 30%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/padding_plugin.cpp.o
[ 35%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/internal_utils.cpp.o
[ 40%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/transform_plugin.cpp.o
[ 45%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/conv3d_plugin.cpp.o
[ 50%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/softargmax_plugin.cpp.o
[ 55%] Building CXX object lib/CMakeFiles/nvstereo_inference_debug.dir/cost_volume_plugin.cpp.o
[ 60%] Linking CXX static library libnvstereo_inference_debug.a
[ 60%] Built target nvstereo_inference_debug
Scanning dependencies of target nvstereo_sample_app_debug
[ 65%] Building CXX object sample_app/CMakeFiles/nvstereo_sample_app_debug.dir/resnet18_1025x321_net.cpp.o
[ 70%] Building CXX object sample_app/CMakeFiles/nvstereo_sample_app_debug.dir/resnet18_2D_513x257_net.cpp.o
[ 75%] Building CXX object sample_app/CMakeFiles/nvstereo_sample_app_debug.dir/main.cpp.o
[ 80%] Building CXX object sample_app/CMakeFiles/nvstereo_sample_app_debug.dir/nvsmall_1025x321_net.cpp.o
[ 85%] Building CXX object sample_app/CMakeFiles/nvstereo_sample_app_debug.dir/nvtiny_513x161_net.cpp.o
[ 90%] Linking CXX executable ../../bin/nvstereo_sample_app_debug
[ 90%] Built target nvstereo_sample_app_debug
Scanning dependencies of target nvstereo_tests_debug
[ 95%] Building CXX object tests/CMakeFiles/nvstereo_tests_debug.dir/tests_main.cpp.o
[100%] Linking CXX executable ../../bin/nvstereo_tests_debug
[100%] Built target nvstereo_tests_debug
```

rahulsharma11, 2018-07-31:
Ok. Thanks. Will check myself.

closed 2018-07-31 by rahulsharma11