NVIDIA-AI-IOT/redtail #75: Should I modify redtail ros packages or create new packages ?

earcz, 2018-07-26:
Hi,
To implement redtail project by adding new features, I am working on ROS for a while, but got stuck. I want that the drone flies along the x and y directions (like `_altitude_gain:=2`)  for searching a specific object (i.e maze) after gaining specified altitude. If we split the tasks into 2

1) For movement along the x and y directions, how can I add the feature of flying along x and y directions?  Which ros files of redtail should I modify? or Should I create new files? 

2) For searching specific object (idk if YOLO detects maze or not), I need to use image processing. Is there any easy way to do this by using ROS? (i.e readily available ros packages) 

Best.

Alexey-Kamenev, 2018-07-30:
There are few things you may want to consider.

1. If you goal is to simply move laterally, then you can use the same approach we used in our `px4_controller` which is a waypoint navigation via MAVROS. To move the drone in x,y directions, `px4_controller` publishes waypoints using [setpoint_position](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_position) topic. Please refer to MAVROS wiki and `px4_controller` sources for more details.

2. If your goal is to perform complex navigation using data from multiple sensors then you should look at [ROS navigation](http://wiki.ros.org/navigation) stack.

3. Our YOLO DNN was trained on PASCAL VOC dataset so it recognizes only simple classes from that dataset. If you want something else, you can either train a different DNN or use some other computer vision techniques (e.g. from OpenCV). ROS provides integration with OpenCV via [vision_opencv](http://wiki.ros.org/vision_opencv) package.

earcz, 2018-07-31:
Thank you Alexey, you provided good points.
One more thing. I am trying to change the location of the host computer for qgroundcontrol. Should I modify the `redtail/ros/packages/px4_controller/launch/mavros_caffe_ctl.launch` for `fcu_url` and `gcs_url`?

Alexey-Kamenev, 2018-08-01:
`fcu_url` parameter specifies the address of FCU (e.g. real Pixhawk or the simulator). `gcs_url` specifies the address of a machine where GCS (Ground Control Station, e.g. QGroundControl) is running. See [this](http://wiki.ros.org/mavros#Connection_URL) for more details.
As for the map in QGC - since you don't have a GPS lock, it will use default map which is Zurich. You can probably change that to something else but I haven't tried that. Hopefully QGC documentation can provide more details.

earcz, 2018-08-02:
Thank you for the information.

earcz, 2018-08-03:
I needed to reopen the topic. You said that I don't have GPS lock. However, when I start the gazebo simulator with mavros and QGroundControl, there is written `GPS lock: 3D Lock` under the GPS status toolbar. Doesn't that mean that I have a GPS lock ? 

[image]

closed 2018-09-11 by earcz