NVIDIA-AI-IOT/redtail #78: End to End Tests fail with timeout

abarajithan11, 2018-08-16:
I am using a Jetson TX2 board with Jetpack 3.2.1 and L4T 28.2.

I cloned the repo into ~/redtail, changed variables accordingly, merged the yolo-relu models and ran the following end to end tests for the DNN:

```
export REDTAIL_TEST_DIR=~/redtail/ros/packages/caffe_ros/tests/data
export REDTAIL_MODEL_DIR=~/redtail/models/pretrained
rostest caffe_ros tests_basic.launch test_data_dir:=$REDTAIL_TEST_DIR \ trail_prototxt_path:=$REDTAIL_MODEL_DIR/TrailNet_SResNet-18.prototxt trail_model_path:=$REDTAIL_MODEL_DIR/TrailNet_SResNet-18.caffemodel \ object_prototxt_path:=$REDTAIL_MODEL_DIR/yolo-relu.prototxt object_model_path:=$REDTAIL_MODEL_DIR/yolo-relu.caffemodel
```

But I get the following error:

[image]

Here is my log file:

[rostest-tegra-ubuntu-4835.log](https://github.com/NVIDIA-Jetson/redtail/files/2292068/rostest-tegra-ubuntu-4835.log)

Kindly help me with this issue.

Alexey-Kamenev, 2018-08-16:
If you are running it on Jetson, it might be timing out as compilation takes several minutes. Try increasing the timeout or switch Jetson to Max-N mode:
```
sudo nvpmodel -m 0
```

abarajithan11, 2018-08-17:
Thanks. I tried it now and it also fails with the same results.