NVIDIA-AI-IOT/redtail #82: Breakout Board

CapitalZe, 2018-08-30:
I had asked the question before, but I probably posed the question wrong.

I am trying to source the 8 channel PWM breakout board photographed in the wiki.
I have attached a screenshot of the component I am referring to.

This one seems to have a 10-pin JST input. I am unable to find a manufacturer, part number or anywhere that supplies this particular unit.

If anyone could give me any advice, guidance or a pointer I'd be very grateful. It is literally the last component I am waiting on, I have everything else ready and at hand.

Many Thanks!

[image]

Alexey-Kamenev, 2018-08-30:
Sorry, I misunderstood your question the first time, for some reason I thought the question was about receiver/transmitter.

As for the breakout board: as I remember, we ordered parts directly from TBS as an ARF kit with few customizations that included replacing DJI Naza with PixFalcon (unfortunately, TBS does not sell these kits anymore). The breakout board came with the kit. As I understand, the breakout board is actually a part of [Pixhawk Mini](https://github.com/PX4/px4_user_guide/blob/master/en/flight_controller/pixhawk_mini.md#pixhawk-mini-vs-pixfalcon) not PixFalcon but apparently TBS included it as well. If you don't have it then you can use PWM breakout cable that comes with [PixFalcon](http://www.holybro.com/product/8). The board is needed only when you need to power PWM rail which you don't need to do in case of the drone.
Hope this helps.

CapitalZe, 2018-09-05:
Thanks very much! You're time and effort really helped.

closed 2018-09-05 by CapitalZe