NVIDIA-AI-IOT/redtail #84: Detected 0 radio channels. To operate PX4, you need at least 5 channels

CapitalZe, 2018-09-12:
So this is probably more of a drone hobbyist issue and purely redtail.

I'll post my question anyway, incase someone else has gone through it and can help, thereafter also potentially helping anyone who may face it in the future.

I have conducted a succesful bind between my d4r-ii Rx and Taranis X9D Plus Tx.
The d4r-ii Rx is connected correctly to my PixFalcon.
I am unable to complete QGroundControl calibration and move on to testing out the pretrained networks because everytime I attempt to calibration the Radio in QGroundControl I am met with the message:

Detected 0 radio channels. To operate PX4, you need at least 5 channels

I have no response when trying to calibrate other than the message above.
There is very little information about this online, I've found less than a dozen pages dealing with it and none of them help me solve this issue.

If anyone has any input, I'd be very grateful. I've been curtailed by this for many days now.

Alexey-Kamenev, 2018-09-12:
How did you test that the bind was successful? For example, when you turn on both Taranis and PixFalcon with D4R, does D4R LED turn green? If there are 0 channels detected that means that either bind was not successful or something is turned off/not powered properly.

CapitalZe, 2018-09-12:
The D4R light turns green and the Taranis informs me that: "Telemetry Recovered".
If I switch off the Taranis then the light on the D4R turns red. Alternatively, if I disengage the power from the D4R the Taranis informs me: "Telemetry Lost".
Then, when power is regained Taranis informs me again that: "Telemetry Recovered".

One thing I have read is that maybe the USB is not providing enough power and I should use 5v power source to seperately power the D4R.
I've tried multiple firmware on the PixFalcon (knowing that you specify the 'px4fmu-v2_lpe.px4'), but nothing works.

I'm beginning to think I have a faulty unit.

Any ideas?

Thanks for your time!

CapitalZe, 2018-09-13:
I solved it.

It was a simple matter, after all.

I didn't jump the terminals on the d4r-ii when originally binding the Tx and Rx. I misplaced the jumper hubs that came with it but a makeshift one sufficed.

Cheers!

closed 2018-09-13 by CapitalZe