NVIDIA-AI-IOT/redtail #85: Jetson Wi-Fi Access Point is not working outside

earcz, 2018-09-12:
Just before flying, I created a wifi acces point. When I am at the office, I can connect to this AP without any problem. However, when I go to the field to launch ros nodes and fly the drone, I cannot connect to the access point through wifi. I have tried that I connected to AP in the office and went outside, this time it disconnected.

I thought that the problem is related to GPS and it corrupts the wifi network. However, I can also have nearly 15 GPS in indoor like outdoor. I could not figure out how to solve this problem for a while. Why can't I connect to jetson+orbitty when outside ?

Alexey-Kamenev, 2018-09-13:
I don't remember having that particular problem in our case. In the open field, WiFi range was about 50-100 meters depending on which WiFi antenna we used in our laptop (i.e. internal vs. external USB high-gain). In the forest it was about 20-30 meters.
We had some issues with USB3.0 connection (e.g. ZED camera) creating interference which affected PixFalcon/GPS but it was fixed after we shielded it properly.

earcz, 2018-09-13:
I am suffering from this interesting problem and cannot fix it. Even my pc and jetson are nearby (outdoor  environment), I cannot ssh into it. Thus, range is not the issue I guess. Can you provide any suggestion with respect to your knowledge ?

Alexey-Kamenev, 2018-09-26:
One thing that you can try is to connect Orbitty to your laptop via Ethernet cable, go outside, ssh using Ethernet IP and check/debug WiFi interface/logs/etc.

closed 2018-09-26 by Alexey-Kamenev