NVIDIA-AI-IOT/redtail #88: Slower Runtime of StereoDNN

raghul9706, 2018-09-14:
I built stereoDNN on Jetson Tx2 board with CUDA toolkit 9.0. When I ran the nvtiny format with 513x161 Input size, the Host time was around 400 ms.The application took nearly 10 seconds for loading the left and right images ( the ones given in the "data" folder ) and to save the output png file. Also, I tried to stream from a live camera using a ros wrapper written by me and the update rate of Disparity was around 1fps.

Is it usual or am I missing something to lessen the time of run? 

**Input**
```
$ ./nvstereo_sample_app nvsmall 513 161 ../../models/NVTiny/TensorRT/trt_weights.bin img_left.png img_right.png ./disp.bin
```
**Output**
(Omitting the logger messages)
```
Using fp32 data type.
Loaded 48 weight sets.
Using [513, 161](width, height) as network input dimensions.
Host time: 434.7779ms
Done
```

**Environment details**
_OS_: Ubuntu 16.04
_Board_: NVIDIA Jetson TX2
_CUDA_: 9.0
_Kernel version_: 4.4.38-tegra

Alexey-Kamenev, 2018-09-14:
The reason for such long loading time is TensorRT graph optimizer performing runtime analysis and optimization of the graph (model). The optimizer produces optimized execution plan which can then be serialized on disk to speed up loading time on subsequent runs, however, Stereo DNN library currently does not support such serialization. We will add support for serialization and enable faster loading in the next release (very soon!).

As for inference time of the model itself: the timings provided are for TX2 in MAX-N + max clocks mode. To switch Jetson to this mode, run the following:
```
sudo nvpmodel -m 0
sudo ~/jetson_clocks.sh
```
And run the sample again.

Alexey-Kamenev, 2018-09-19:
Update: I've just [merged](../pull/90) the changes which improve runtime performance and reduce model load time. Note that this is only for `ResNet 18 2D` model.

Alexey-Kamenev, 2018-09-24:
Closing the issue, feel free to re-open or create a new one.

closed 2018-09-24 by Alexey-Kamenev