NVIDIA-AI-IOT/redtail #89: Zed Launch

CapitalZe, 2018-09-19:
Before I make any changes anywhere I'd like to confirm what I infer.
Referring to [this](https://github.com/NVIDIA-Jetson/redtail/issues/76) issue.

I have a ZED and am currently getting the error:
```
[ INFO] [1537347897.712051688]: camera calibration URL: file:///home/nvidia/.ros/camera_info/camera.yaml
[ INFO] [1537347897.712238568]: Unable to open camera calibration file [/home/nvidia/.ros/camera_info/camera.yaml]
[ WARN] [1537347897.712288520]: Camera calibration file /home/nvidia/.ros/camera_info/camera.yaml not found.
[ INFO] [1537347897.712326440]: Loaded camera calibration from 
[ WARN] [1537347897.713582248]: No camera frame_id set, using frame "/camera_frame".
[ INFO] [1537347897.756565640]: Time offset: 1537347759.181
[FATAL] [1537347897.756848360]: Failed to PAUSE stream, check your gstreamer configuration.
[FATAL] [1537347897.756897384]: Failed to initialize gscam stream!
```

From reading the above linked issue, it apparent that I didn't launch the ros-zed-wrapper node.

So my question is, must I emend the everything.launch file to remove the gscam input and replace it with [this](https://github.com/NVIDIA-Jetson/redtail/blob/master/ros/packages/redtail_debug/launch/trailnet_debug_zed.launch) or must I simply change the pertinent topic of the caffe_ros.cpp node?

Cheers!

Alexey-Kamenev, 2018-09-19:
That's right, if you use ZED ROS node then you don't need to run `gscam` node. In such case you can create your own .launch file, based on `trailnet_debug_zed.launch` you referenced. You don't need to change the source code, `caffe_ros` ROS node has a parameter, `camera_topic` which can be used to change camera topic when running `caffe_ros` via `rosrun` or .launch file.

CapitalZe, 2018-09-22:
Thanks! Really helpful. As of now it's working.

closed 2018-09-22 by CapitalZe