NVIDIA-AI-IOT/redtail #91: Error building docker simulation from Matplotlib 3.0+

kidtronnix, 2018-09-20:
When I run...

```
./build_redtail_image.sh TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-9.0.cudnn7.0.tar.gz
```

I get the following error...

```
Collecting px4tools
  Downloading https://files.pythonhosted.org/packages/bc/95/bdd2577e99e03a991d7eb5e507c746a624914dc4b0f0feb6623a0956a81e/px4tools-0.9.6.tar.gz (42kB)
Collecting control>=0.7.0 (from px4tools)
  Downloading https://files.pythonhosted.org/packages/89/6b/5ab127da1f93d8f12d108194b0ae4d245c77b97aa57ac01e9200a70d15b9/control-0.8.0.tar.gz (165kB)
Collecting matplotlib>=2.2.2 (from px4tools)
  Downloading https://files.pythonhosted.org/packages/ec/06/def4fb2620cbe671ba0cb6462cbd8653fbffa4acd87d6d572659e7c71c13/matplotlib-3.0.0.tar.gz (36.3MB)
    Complete output from command python setup.py egg_info:
    
    Matplotlib 3.0+ does not support Python 2.x, 3.0, 3.1, 3.2, 3.3, or 3.4.
    Beginning with Matplotlib 3.0, Python 3.5 and above is required.
    
    This may be due to an out of date pip.
    
    Make sure you have pip >= 9.0.1.
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-zE8GD6/matplotlib/
You are using pip version 8.1.1, however version 18.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The command '/bin/sh -c wget --quiet http://packages.osrfoundation.org/gazebo.key -O - | apt-key add -     && sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial main" > /etc/apt/sources.list.d/gazebo-stable.list'     && apt-get update && apt-get -y --no-install-recommends install         gazebo7                 libgazebo7-dev      && pip install px4tools     && git clone https://github.com/dronekit/dronekit-python.git                             && (cd dronekit-python && pip install -r requirements.txt) && rm -rf dronekit-python     && git clone https://github.com/ArduPilot/pymavlink.git && cd pymavlink                                 && git clone git://github.com/mavlink/mavlink.git && ln -s ${PWD}/mavlink/message_definitions ../       && pip install . && cd .. && rm -rf pymavlink && rm -rf message_definitions' returned a non-zero code: 1
```

Any help is appreciated. I think we just need to lock down a specific `matplotlib`

kidtronnix, 2018-09-21:
Found the fix, simply upgrade pip in the build.

Alexey-Kamenev, 2018-09-21:
Thank you for reporting the issue and creating PR!
The PR as it is, unfortunately, will break the Docker as the added line is not a valid Docker syntax. There should be either `RUN` Docker command (e.g. `RUN pip install --upgrade pip`) or the command should be rolled into previous `RUN` command.
However, there is a bigger problem: after updating pip, the build script will fail with a different error:
```
raceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name 'main'
```
This is a [known issue](https://stackoverflow.com/questions/49836676/error-after-upgrading-pip-cannot-import-name-main) and the workaround is to avoid upgrading pip.
Ideally, all Python packages should be installed in a virtual environment (e.g. like we do in our [TF container](../tree/master/tools/tensorflow/docker)) but it would be hard to do in the current `redtail` image with lots of dependencies which might not work well in virtualenvs.
So what I did instead is to do exactly what you suggested in the original post: lock matplotlib version. Not ideal, but the image is now building fine and all components seem to work (I did a clean image build and tested it). I've just pushed the fix to master.

Also, you should be using TRT 4.0, not 3.0. Something like that:
```
./build_redtail_image.sh /data/downloads/TensorRT-4.0.1.6.Ubuntu-16.04.4.x86_64-gnu.cuda-9.0.cudnn7.1.tar.gz
```

cookiegg, 2018-09-22:
@Alexey-Kamenev  I got same problem. Could you tell me how to lock matplotlib version? Thanks!

cookiegg, 2018-09-22:
> @Alexey-Kamenev I got same problem. Could you tell me how to lock matplotlib version? Thanks!

Finally, I find the master branch 'Dockerfile.kinetic' added 
```shell
&& pip install 'matplotlib==2.2.2' --force-reinstall 
```
The problem has been solved.

Alexey-Kamenev, 2018-09-24:
Closing the issue, feel free to re-open or create a new one.

closed 2018-09-24 by Alexey-Kamenev

kidtronnix, 2018-09-24:
Huh, sorry for the incorrect syntax. I just added the fix after trying it on my side and simply forgot to add the RUN.

The pip upgrade didn't break things for me, but locking matplotlib seems to work too :)