NVIDIA-AI-IOT/redtail #94: Image Encoding bgra8

CapitalZe, 2018-09-25:
I'm able to launch ZED and everything else I require, but two issues arise which result in the trails_dnn-4 finishing cleanly.

`[ INFO] [1537795203.312240992]: Advertised on topic path_odom
[ INFO] [1537795203.313778048]: Advertised on topic path_map
[ WARN] [1537795203.313983104]: 'imu_pub_rate' set to 500 Hz but ZED camera model does not support IMU data publishing.
[ INFO] [1537795203.442537792]: Starting Tracking
[ INFO] [1537795203.448557440]: Init Odometry with first valid pose data : 1
[ INFO] [1537795203.448726304]: Pose Smoothing : 0
[ INFO] [1537795203.448779168]: Spatial Memory : 0
[ INFO] [1537795203.448819456]: Floor Alignment : 1
[ INFO] [1537795203.521931232]: Tracking ENABLED
[FATAL] [1537795203.571486656]: Image encoding bgra8 is not yet supported. Supported encodings: rgb8, bgr8
[trails_dnn-4] process has finished cleanly
log file: /home/nvidia/.ros/log/88d5fa64-bffc-11e8-9f2b-00044b8d7ffe/trails_dnn-4*.log
`
I've been scouring through and can't seem to find where I can change the encoding from bgra8 to rgb8 or bgr8. I assume its on the ZED side of things.

Cheers!

Alexey-Kamenev, 2018-09-25:
This was recently changed on ZED ROS side when they [removed dependency on OpenCV](https://github.com/stereolabs/zed-ros-wrapper/pull/272). That CL has removed [RGBA->RGB](https://github.com/stereolabs/zed-ros-wrapper/pull/272/files#diff-927cff144408b0f134adbf9a0d18c772L1426) conversion. 

I'm going to fix (soon) `caffe_ros` node to accept `bgra8` encoding as well.

closed 2018-09-25 by Alexey-Kamenev

CapitalZe, 2018-09-26:
That was quick! Thanks very much.

commit 08c53f73 (2018-09-25): Added support for BGRA image topics (ZED). Fixes #94.
commit 2bfe34fa (2018-09-25): Merge pull request #95 from NVIDIA-Jetson/alexeyk/bgra

Added support for BGRA image topics (ZED). Fixes #94.