NVIDIA-AI-IOT/redtail #97: QGroundControl - FCU issue

CapitalZe, 2018-09-28:
According to rqt_graph eveything is about right.

[image]
{the rqt_graph is on the Jetson and it was easier to just take a pic rather than a screenshot}

I am unable to arm my Px4 however.

```
[ INFO] [1537969550.629318456]: Pos:0.00, 0.01, -1.28, Att:177.15, 175.32, 11.30, SetAlt:-1.33
[ INFO] [1537969551.679325592]: Pos:0.00, 0.01, -1.28, Att:177.15, 175.32, 11.26, SetAlt:-1.33
[ INFO] [1537969552.729309048]: Pos:0.00, 0.01, -1.28, Att:177.16, 175.31, 11.26, SetAlt:-1.33
[ INFO] [1537969553.779309656]: Pos:0.00, 0.01, -1.28, Att:177.16, 175.31, 11.22, SetAlt:-1.33
[ INFO] [1537969554.829335256]: Pos:0.00, 0.01, -1.27, Att:177.16, 175.31, 11.22, SetAlt:-1.33
[ INFO] [1537969555.879328824]: Pos:0.00, 0.01, -1.27, Att:177.15, 175.30, 11.24, SetAlt:-1.33
[ERROR] [1537969556.129258616]: Could not arm PX4/FCU!
```

Now this is occurring because my QGC is not connecting to my vehicle over the WiFi AP.
My host computer connects fine. I am using VMware on a Mac (not sure if this is a source of problem), and I can establish an SSH and launch everything I need to.

Everything launches and connects but after a moment or so I am faced with the above error code and all the while my QGC (hosted on Ubuntu 16.04 in VMWare) does not connect with my vehicle stating in red "waiting for vehicle connection".

I have used:

```
roslaunch mavros px4.launch fcu_url:="/dev/ttyTHS1:921600" gcs_url:="udp://@10.42.0.85"
```

To separatley test if Mavros is working etc, adjusting the IP to my hosts IP, and no progress.

Has anyone faced this, or have any idea on what the problem could be?

Alexey-Kamenev, 2018-09-28:
Is the IP address of your GCS machine really `10.42.0.85`? You can get the IP using `ifconfig` command and update `gcs_url` parameter accordingly.

CapitalZe, 2018-09-29:
What I meant to say is, I changed the 10.42.0.85 to my IP.
My IP, as you predicted is, 10.42.0.xxx = 10.42.0.121.

Now, this is of my Mac, which is running Ubuntu via VMWare, which itself has a reported IP of 192.168.212.131 via the 'wired connection' used to communicate to the Mac.
Neither work, and I'm wondering if using a virtual machine is the problem.

I'll get a dedicated linux machine and try with that, it very well may be the case.

I will update with any info I find, and see what I can do about QGC.

Cheers!

CapitalZe, 2018-10-01:
I got it to work.

Ultimately it was a loose wire from the Px4 to an ESC. It was in the right place, but sightly loose, and seeing as I don't have a breakout board and wired directly, it was kind of difficult to spot.

As for anyone else using a Virtual Machine for Ubuntu host, you must use the IP of the actual machine, not the one reported in the Virtual Machine.
I hope this helps.

Cheers!

closed 2018-10-01 by CapitalZe