NVIDIA-AI-IOT/redtail #99: StereoDNN Build Error

mohanen, 2018-10-12:
> This is my cmake output 

```
-- Found CUDA: /usr/local/cuda (found version "10.0") 
-- Found OpenCV: /usr/local/opencv (found version "3.4.2") 
-- Found CUDA: /usr/local/cuda (found suitable version "10.0", minimum required is "9.0") 
-- Found OpenCV: /usr/local/opencv (found suitable version "3.4.2", minimum required is "3.3.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mohanen/mohanen/vision/ml/redtail/stereoDNN/build
[  5%] Building CXX object lib/CMakeFiles/nvstereo_inference.dir/elu_plugin.cpp.o
/home/mohanen/mohanen/vision/ml/redtail/stereoDNN/lib/elu_plugin.cpp: In member function ‘virtual nvinfer1::IPlugin* redtail::tensorrt::PluginContainer::createEluPlugin(nvinfer1::DataType, std::__cxx11::string)’:
/home/mohanen/mohanen/vision/ml/redtail/stereoDNN/lib/elu_plugin.cpp:219:59: error: invalid new-expression of abstract class type ‘redtail::tensorrt::EluPlugin’
     plugins_.push_back(new EluPlugin(data_type, log_, name));
                                                           ^
/home/mohanen/mohanen/vision/ml/redtail/stereoDNN/lib/elu_plugin.cpp:16:7: note:   because the following virtual functions are pure within ‘redtail::tensorrt::EluPlugin’:
 class EluPlugin: public IPluginExt
       ^
In file included from /home/mohanen/mohanen/vision/ml/redtail/stereoDNN/lib/internal_utils.h:14:0,
                 from /home/mohanen/mohanen/vision/ml/redtail/stereoDNN/lib/elu_plugin.cpp:4:
/usr/include/x86_64-linux-gnu/NvInfer.h:2660:25: note: 	virtual const char* nvinfer1::IPluginExt::getPluginType() const
     virtual const char* getPluginType() const = 0;
                         ^
/usr/include/x86_64-linux-gnu/NvInfer.h:2666:25: note: 	virtual const char* nvinfer1::IPluginExt::getPluginVersion() const
     virtual const char* getPluginVersion() const = 0;
                         ^
/usr/include/x86_64-linux-gnu/NvInfer.h:2671:18: note: 	virtual void nvinfer1::IPluginExt::destroy()
     virtual void destroy() = 0;
                  ^
/usr/include/x86_64-linux-gnu/NvInfer.h:2676:25: note: 	virtual nvinfer1::IPluginExt* nvinfer1::IPluginExt::clone() const
     virtual IPluginExt* clone() const = 0;
                         ^
/home/mohanen/mohanen/vision/ml/redtail/stereoDNN/lib/elu_plugin.cpp: In member function ‘virtual nvinfer1::IPlugin* redtail::tensorrt::PluginContainer::deserializeEluPlugin(const char*, const void*, size_t)’:
/home/mohanen/mohanen/vision/ml/redtail/stereoDNN/lib/elu_plugin.cpp:227:60: error: invalid new-expression of abstract class type ‘redtail::tensorrt::EluPlugin’
     plugins_.push_back(new EluPlugin(name, data, size, log_));
                                                            ^
lib/CMakeFiles/nvstereo_inference.dir/build.make:394: recipe for target 'lib/CMakeFiles/nvstereo_inference.dir/elu_plugin.cpp.o' failed
make[2]: *** [lib/CMakeFiles/nvstereo_inference.dir/elu_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'lib/CMakeFiles/nvstereo_inference.dir/all' failed
make[1]: *** [lib/CMakeFiles/nvstereo_inference.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

Alexey-Kamenev, 2018-10-12:
Stereo DNN currently supports CUDA 9 and TensorRT 4. There were some breaking changes introduced in TRT 5, we haven't had time to update the code yet.

mohanen, 2018-10-16:
with tensorrt 4 and cuda 9.2 i was able to run the code

closed 2018-10-16 by Alexey-Kamenev